ARMmbed/mbed-os #12503: LwIP bind() does not promote error to caller

JojoS62, 2020-02-24:
### Description of defect
I'm trying to use async NetworkInterface features, there I ran into a problem when after an interface disconnect/connect a tcp server is not working. A part of the problem is that bind() does not report an error, it is not passed to the caller:
https://github.com/ARMmbed/mbed-os/blob/bb48fa46cdaa583eeef24d128013aab39a564c81/features/lwipstack/lwip/src/api/lwip_api_lib.c#L131-L133

fn() is calling https://github.com/ARMmbed/mbed-os/blob/bb48fa46cdaa583eeef24d128013aab39a564c81/features/lwipstack/lwip/src/api/lwip_api_msg.c#L1216
and setting the err in ```msg->err = err;```, so line133 should return msg->err.

My testcode and the procedure is reported in the forum:
https://forums.mbed.com/t/network-interface-async-handling-tcp-server-restart-problem/7350

#### Target(s) affected by this defect ?

targets that use ethernet with lwip

#### Toolchain(s) (name and version) displaying this defect ?

gcc 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ?

0xc0170, 2020-03-06:
cc @ARMmbed/mbed-os-ipcore

michalpasztamobica, 2020-03-25:
Hi @JojoS62 . Thanks for the report. I checked your forum post and it seems this bug report is actually intended behavior. Or does the part of the problem described above still need investigation?

JojoS62, 2020-03-25:
I think the problem is present, but it will only appear in this case when the socket is reused. When you use the sequence bind() - listen(), then the error is reported in the listen() function, but the bind() failed already. So the return apimsg->err; after calling the fn(apimsg); will fix this.

michalpasztamobica, 2020-03-25:
One problem is we can't tell if the structure was processed correctly. 
The `fn()` f[unction returns void](https://github.com/ARMmbed/mbed-os/blob/bb48fa46cdaa583eeef24d128013aab39a564c81/features/lwipstack/lwip/src/include/lwip/tcpip.h#L72), so there is no way to tell if the apimsg structured was filled in with valid data (like it is for `tcpip_send_msg_wait_sem` a few lines below). Perhaps that's why propagating the error is being skipped?

JojoS62, 2020-03-25:
this is the original lwip code:

```
static err_t
netconn_apimsg(tcpip_callback_fn fn, struct api_msg *apimsg)
{
  err_t err;

#ifdef LWIP_DEBUG
  /* catch functions that don't set err */
  apimsg->err = ERR_VAL;
#endif /* LWIP_DEBUG */

#if LWIP_NETCONN_SEM_PER_THREAD
  apimsg->op_completed_sem = LWIP_NETCONN_THREAD_SEM_GET();
#endif /* LWIP_NETCONN_SEM_PER_THREAD */

  err = tcpip_send_msg_wait_sem(fn, apimsg, LWIP_API_MSG_SEM(apimsg));
  if (err == ERR_OK) {
    return apimsg->err;
  }
  return err;
}
```

the apimsg->err is prepared only in the debug build. So I read this like fn() always has to set the err code. How about always setting the err before calling fn()? If there is a fn that does set the correct return value, the tests should find it?

michalpasztamobica, 2020-03-25:
You have a point. They also assume `fn()` will always set the error code. So returning `apimsg->err` should be fine. As this is indeed mbed's addition to lwip I will prepare a PR with a fix. Thanks for your report once again!

JojoS62, 2020-03-25:
perfect, thanks also!

closed 2020-03-27 by adbridge