ARMmbed/mbed-os #12504: GCC: Callee-saved registers corrupted in TZ non-secure entry function

ccli8, 2020-02-25:
# Description of defect

According to [5.4.2 of ARM v8-M Security Extensions: Requirements on Development Tools](https://developer.arm.com/docs/ecm0359818/11), on return from an non-secure entry function, callee-saved registers as mandated by AAPCS should be restored, but they are incorrectly cleared in GCC-compiled code.

#### Target(s) affected by this defect ?

NU_PFM_M2351_* and possibly other TZ targets

#### Toolchain(s) (name and version) displaying this defect ?

GCC 9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

8c1727030607e1a829b0b5023e66294397e6b47a

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli: 1.10.1

#### How is this defect reproduced ? 

**Compile secure code on NU_PFM_M2351_NPSA_S**

1.  Close NuMaker-mbed-TZ-secure-example repository.
    ```sh
    $ git clone https://github.com/OpenNuvoton/NuMaker-mbed-TZ-secure-example -b nuvoton_debug_gcc_nsc
    ```

1.  Navigate to NuMaker-mbed-TZ-secure-example directory. Make sure we have switched to nuvoton_debug_gcc_nsc branch.

    ```sh
    $ cd NuMaker-mbed-TZ-secure-example
    ```

1.  Enable GCC build on NU_PFM_M2351_NPSA_S. By default, GCC is disabled due to this issue.
    https://github.com/ARMmbed/mbed-os/blob/1629103fb1bd66ba4591d78bc4faf74234c33e99/targets/targets.json#L13120-L13131

1.  Build on NU_PFM_M2351_NPSA_S/GCC_ARM

    ```sh
    $ mbed compile -m NU_PFM_M2351_NPSA_S -t GCC_ARM --app-config mbed_app_custom.json
    ```

1.  Keep secure lib/image for later use

    BUILD/NU_PFM_M2351_NPSA_S/GCC_ARM/cmse_lib.o
    BUILD/NU_PFM_M2351_NPSA_S/GCC_ARM/NuMaker-mbed-TZ-secure-example.hex

**Compile non-secure code on NU_PFM_M2351_NPSA_NS, against secure lib/image built above**

1.  Clone NuMaker-mbed-TZ-nonsecure-example repository.

     ```sh
    $ git clone https://github.com/OpenNuvoton/NuMaker-mbed-TZ-nonsecure-example -b nuvoton_debug_gcc_nsc
    ```

1.  Navigate to NuMaker-mbed-TZ-nonsecure-example directory. Make sure we have switched to nuvoton_debug_gcc_nsc branch.

    ```sh
    $ cd NuMaker-mbed-TZ-nonsecure-example
    ```

1.  Create TARGET_NU_PFM_M2351_NPSA_NS directory. Copy secure lib/image built above here.

1.  Build on NU_PFM_M2351_NPSA_NS/ARMC6. ARMC6 can replace with GCC_ARM or IAR. For non-secure build, toolchain selection doesn't matter for this issue.

    ```sh
    $ mbed compile -m NU_PFM_M2351_NPSA_NS -t ARMC6 --app-config mbed_app_custom.json
    ```

1.  Keep built image, which has combined non-secure and secure images

    BUILD/NU_PFM_M2351_NPSA_NS/ARMC6/NuMaker-mbed-TZ-nonsecure-example.hex

**Execution result**

1. Configure terminal program with 115200/8-N-1

1. Flash the above image onto M2351 board

    BUILD/NU_PFM_M2351_NPSA_NS/ARMC6/NuMaker-mbed-TZ-nonsecure-example.hex

1. We can see callee-saved register R8 is corrupted in non-secure entry function

    ```
    Check if called-saved registers are corrupted across NSC call...
    if we see error message or see nothing below, it means called-saved registers have been corrupted across NSC call!!!
    R8(before NSC)=0x12345678, R8(after NSC)=0x00000000
    ERROR: Callee-saved R8 corrupted in NSC
    ```

1. If we re-produce the steps above, but change secure build toolchain to ARMC6 from GCC_ARM, we can see callee-saved register R8 is not corrupted.

    ```
    Check if called-saved registers are corrupted across NSC call...
    if we see error message or see nothing below, it means called-saved registers have been corrupted across NSC call!!!
    R8(before NSC)=0x12345678, R8(after NSC)=0x12345678
    OK: Callee-saved R8 not corrupted in NSC
    ```

MarceloSalazar, 2020-02-25:
Note: related to https://jira.arm.com/browse/GNUTOOLS-9066

0xc0170, 2020-02-25:
> Note: related to https://jira.arm.com/browse/GNUTOOLS-9066

Thanks for the reference. @ccli8 We will keep you updated (the reference is an internal ticket).

closed 2021-10-20 by ciarmcom