ARMmbed/mbed-os #12532: Unexpected results with I2C API on MAX32630FTHR

idea--list, 2020-02-28:
### Description of defect
Found a useful I2C scanner for Arduino [here](https://playground.arduino.cc/Main/I2cScanner/).
As i could not find similar one for mBed, i tried to code it myself.
During the implementation i faced 2 different possible bugs.

bug 1: if i try to scan the ports lower than address = 10, then the scan result is all the way corrupt, however if i start at address = 10 the scan finds both peripheral addresses this target has on-board (MAX14690 at 0x28 and BMI160 at 0x68). I have no clue why this happens and how to counter act it?

bug 2: Though i did not reviewed it's code yet, the stop() method of the I2C API behaves in an unexpected way for me: if it is placed within the loop, then the result of the scan becomes unconsistent:
- In the first cycle the scan will find 2 devices as expected
- In the second cycle the scan will find 5 devices, which is wrong
- In the third cycle and beyond the scan finds 0 devices

Can you please confirm these?
Please also check on other targets as i only have this board.

#### Target(s) affected by this defect ?
MAX32630FTHR (this is my only mbed capable board, so no clue how other targets behave)

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6

#### What version of Mbed-os are you using (tag or sha) ?

5.14.2, 5.15.0, 5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio V0.8.1

#### How is this defect reproduced ? 
Compile this code:
`#include "mbed.h"

I2C i2c(I2C2_SDA, I2C2_SCL);
Serial pc(USBTX, USBRX);

int address;
char dummybyte[0];
int result;
int nDevices;

int main() {
  dummybyte[0] = 0x00;

  while (1) {
    i2c.start();        //is ok if used in the loop
    nDevices = 0;

    for (address = 10; address < 128; address++) {      // addresses less than 10 cause bug #1

      result = i2c.write(address << 1, dummybyte, 1);
      printf("Result at 0x%X (%i): %s\n", address, address,
             (result == 0 ? "ACK" : "NAK"));
      if (result == 0) {
        nDevices++;
      }
    }

    pc.printf("Devices found: %i\n\n", nDevices);
    ThisThread::sleep_for(4000);
    //i2c.stop();      //introduces a bug #2 if placed in the loop
  }
  return 0;
}
`

0xc0170, 2020-02-28:
Looking at docs first - https://os.mbed.com/docs/mbed-os/v5.15/apis/i2c.html . `write` method provides the transaction - including start/stop - there's argument for repeated. In that case, it would not send the stop but repeated start and the last one should stop the transaction.

Another use case, would be using byte write `int I2C::write(int data)` - this one does not send start, stop - you should use `start/stop` methods and make a transaction. This might be confusing, we should make documentation more clear about this - i2c doxygen docs need an update.

In that case, if you use byte write, it should work but I would rather use full transaction functionality.

cc @ARMmbed/mbed-os-hal 

>  it's documentation is absolutely useless

To improve, please be more specific. I reviewed start/stop and byte write, they do not state their intention , but write/read state its complete transaction. We will fix this.

idea--list, 2020-02-28:
@0xc0170 Thanks for clarifying!

So if i understand it right
- if i use bytewise method int write (int data), then i would need to call .start() and .stop() before and after moving all the data.
- if i use the int write (int address, const char *data, int length, bool repeated=false) then it will know how many bytes i want to move and will handle .start() and .stop() automatically

> To improve, please be more specific. I reviewed start/stop and byte write, they do not state their intention , but write/read state its complete transaction. We will fix this.

If i would be as specific as you want, then i should write a really long list and it would take several days. Exactly because of doxygen just spits out characters that build up sentences...but that does not automatically mean those are understandable for a human being or that it never misses important details. So it is a general issue, IMHO only rewamping the whole doc manually would make a difference. I know that is not an easy task but right now you could easily afford that.

I mean in comparison to Arduino for example Mbed has all the bells and whistles. Exactly that drives many of us to opt for it despite the docs. However recently more and more advanced boards get ported under Arduino and that framework has a much better documentation. But advanced boards will mean that most probably they also will begin to add more advanced features to their framework. If you do not react in the right time, it might lead to a state where Arduino is almost as advanced as Mbed but has far better docs. At that point Mbed would become far less attractive.

0xc0170, 2020-03-02:
>So if i understand it right

Yes, that is how I understand it as well.

Thanks for the feedback, I'll share it with the team

0xc0170, 2020-03-02:
@idea--list don't need to be that specific, just areas you think we can improve (in this case this is targeting drivers doxygen doc, but if there are other areas like our docs portal or other, let us know).

SeppoTakalo, 2020-03-02:
I cannot verify the behaviour, and the code looks incorrect.

Try with this:
```C++
#include "mbed.h"

int main() {
    I2C i2c(I2C_SDA, I2C_SCL);

    for (int i = 1;i < 11; ++i) {
        printf("Scanning I2C devices, loop %d\n", i);
        int n_found = 0;

        for (uint8_t addr = 1; addr < 127; ++addr) {
            i2c.start();
            if (i2c.write(addr << 1) == 1) {
                printf("Found I2C device %02x\n", addr);
                ++n_found;
            }
            i2c.stop();
        }

        printf("Done\nFound %d devices\n", n_found);
        ThisThread::sleep_for(4000);
    }
    return 0;
}

```

For K64F + some shield with I2C devices, shows same number of devices on every loop.
As you can see, you don't need any dummy bytes, or extra writes. I2C behaves the way that after `START` condition the slave returns with `ACK` if there is match in the address, therefore you can immediately do `STOP` condition.

For *mbed_app.json* you might want to use following, so no need to use the `Serial pc(...)`
```
{
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 115200,
            "platform.stdio-convert-newlines": true
        }
    }
}
```

idea--list, 2020-03-03:
@SeppoTakalo 
Thanks for the effort and advice! Tried your code but interestingly on the MAX32630FTHR the code behaves quite different from how you described it on the K64F. Am really confused why this might happen.

On the I2C1 port i only have a GPS module attached at the address 0x10, on I2C2 the board has an on-board PMIC at 0x28 and an accelerometer at 0x68. My original code finds them.
Your code finds 112 devices on the I2C1 and 126 devices on the I2C2 port on the MAX32630FTHR board.

Took a look at the documentation of the I2C API and there i see 2 different write methods:
- `int write (int address, const char *data, int length, bool repeated=false)` this has 4 args including the exact address. This method returns 0 on success.
- `int write (int data)` which takes **data** and not address as argument. My understanding based on the doc is that this data will be written to the bus, but for me it is unclear to which address. This method returns 1 on success.

To sum it up `i2c.write(addr << 1)` in your code definitely returns 1 on my board, but most probably in quite a different context than on your K64F. I am unsure if it might be a target specific bug, however i expect Mbed to behave similarly on whatever board. So i am lost here why all this happens.

idea--list, 2020-03-03:
@0xc0170 
Let me propose to change the current "I2C hello, world" example of the I2C API found [here](https://os.mbed.com/docs/mbed-os/v5.15/apis/i2c.html) to something similar to @SeppoTakalo's code as that is more explanatory as the current example. While one also would not need the exact same temperature sensor. However first we should be sure that the proposed new example code behaves the same way accross all the boards.

Actually i am also wondering if `int write (int data)` might be an undocumented method as in the doc the only I2C::write method with 1 argument has data as argument and not address.

SeppoTakalo, 2020-03-03:
`I2C::write(byte)` is not really undocumented, but the documentation assumes that you have knowledge of how I2C behaves. Basically there is no address line, and therefore the first byte written to the bus, after a START condition, is an address.

[image]

`I2C::write(address, data, length)` just does internally all START and STOP conditions. The Address is just the first byte it writes, then it follows with multiple `write(byte)` calls.

But there might be now a problem that maybe MAX32630FTHR hal driver has internally some clever state machine, and manually writing just one byte, and sending STOP condition, just does not work.

This clearly needs work from @ARMmbed/mbed-os-hal team. Check whether the driver is correctly implemented.

@idea--list in the mean time, can you try replacing the internal loop with:
```C++
for (uint8_t addr = 1; addr < 127; ++addr) {
            char dummy = 0;
            if (i2c.write(addr << 1, &dummy, 1) == 0) {
                printf("Found I2C device %02x\n", addr);
                ++n_found;
            }
        }
```

That is equal to your original example, but without the extra `i2c.start()` which you should not need as the `write()` does a complete transaction.

idea--list, 2020-03-03:
@SeppoTakalo 
Thanks again! I already tried your latest suggestion as it really simplifies my original code. On the MAX32630FTHR it results in:

- Finding 5 devices in the 1st cycle and 0 devices in any further cycles on the I2C1 port which has 2 devices attached.
- If i also modify the for condition to `for (uint8_t addr = 10; addr < 127; ++addr) {` then suddenly the logic changes and it finds 2 devices in each cycle just as expected.

So may be the board is not well-tempered in more than 1 way.

idea--list, 2020-03-13:
Meanwhile i tried to port [this Arduino library](https://github.com/sparkfun/SparkFun_I2C_GPS_Arduino_Library) to mBed.
The GPS module has the address of 0x10 so it is not affected by the bug that i reported here previously.

If i try to run mbed code based on my port of the GPS module, then this code `int ierr = _i2cPort->read(MT333x_ADDR << 1, (char *)&buffer, 255);` returns 1, meaning the read has failed and as a consequence buffer stays empty. This breaks the logic as in the next step i would cycle through each character in that buffer and feed a GPS parser to extract date and time from the raw NMEA sentences.

Interestingly if i compile&run the original Arduino code first and only then compile&run mBed code based on my port my code begins to work properly. But that changes again if i cut the power supply to the board.

After cutting power to the board my ported code begins to fail again until i run the original Arduino library once.

My port might not be perfect, but this puzzles me and i get a feeling as if it was a 3rd type of bug in the I2C API on this target.