ARMmbed/mbed-os #12535: 6.0 Alpha 2, New Error Message for CY Targets:  Bootloader not supported on this target.

ifyall, 2020-02-28:
### Description of defect
In mbed-os-6.0.0-alpha-2 there are two new errors for Cypress targets that are seen when querying the compilation config, compiling or exporting. These two messages are output:

Configuration error: Bootloader not supported on this target. ROM start not found in targets.json.
Configuration error: Bootloader not supported on this target. RAM start not found in targets.json.

These are seen using these commands:
mbed compile --config
mbed compile
mbed export -i eclipse_gcc_arm

#### Target(s) affected by this defect ?
CY8CKIT_062_WIFI_BT
CYW9P62S1_43438EVB_01

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.0.0-alpha-2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
mbed import mbed-os-example-blinky
cd mbed-os-example-blinky/mbed-os
mbed update mbed-os-6.0.0-alpha-2
cd ..
mbed target CYW9P62S1_43438EVB_01
mbed toolchain GCC_ARM
mbed export -i eclipse_gcc_arm
mbed compile --config | grep Bootloader
mbed compile | grep Bootloader

0xc0170, 2020-03-02:
Is that a regression (alpha1 was OK) ?  @ARMmbed/team-cypress Can anyone review and fix?

ifyall, 2020-03-02:
@0xc0170, this is a regression from alpha1. Using the same commands on the same targets does not show this message.

We will do some investigation to determine how this was introduced.

Ian

ifyall, 2020-03-02:
@0xc0170, I have narrowed down the change that caused this, which is https://github.com/ARMmbed/mbed-os/commit/d7e0012bc79d1e94f9e10be5367de2341d25900e. Prior to that commit, the messages were effectively "hidden". It looks like they no longer are hidden and are now standard for all targets, even those without bootloader support.

Thanks,

Ian

0xc0170, 2020-03-06:
Yes, I recall this commit and then some other fixes later. This is a valid error and means target should provide details needed for bootloader. Please review the target configuration and confirm the bootloader is really supported and works (this error means it is not configured properly).

ifyall, 2020-03-06:
@0xc0170,

I think there is some confusion here. The target CYW9P62S1_43438EVB_01 has bootloader_supported = false. I don't think we should complain about it not being configured correctly for the bootloader when it doesn't support the bootloader (yet).

See https://github.com/ARMmbed/mbed-os/blob/mbed-os-6.0.0-alpha-2/targets/targets.json, line 13859.

Ian

0xc0170, 2020-04-02:
I'll check this later today to see if I can reproduce and find the problem.

cc @ARMmbed/mbed-os-tools This would be good to have fixed in 6.0.0-beta-1

0xc0170, 2020-04-02:
The message is misleading.

This is the exception:

```
            return available_memories
        except ConfigException:
            """ If the target doesn't exits in cmsis, but present in targets.json
            with ram and rom start/size defined"""
            start, size = self._get_primary_memory_override(
                active_memory.lower()
            )
            if start is None:
                raise ConfigException(
                    "Bootloader not supported on this target. {} "
                    "start not found in targets.json.".format(active_memory)
                )
            if size is None:
                raise ConfigException(
                    "Bootloader not supported on this target. {} "
                    "size not found in targets.json.".format(active_memory)
                )
            available_memories[active_memory] = [start, size]
            return available_memories
```

What exceptions there state that bootloader can't be supported because some information is missing. This does not have anything related to `bootloader_support`  setting, this comes from the config and cmsis pack. I can patch this error to be more clear, but first need a confirmation that these 2 targets actually contain config + cmsis pack error there. ROM and RAM start are missing (they are `None`), a question is why?

0xc0170, 2020-04-02:
example-blinky, with mbed-os `*   6e46e2ddd3 - (HEAD -> master, upstream/master) Merge pull request #12639 from rajkan01/ethernethal_remove_deprecated (14 hours ago) <Martin Kojtal>
|\` - the latest master as of today.

I run:

```
mbed compile -m CYW9P62S1_43438EVB_01 -t GCC_ARM
```

I dont see any config error output, what have I missed?

0xc0170, 2020-04-02:
As reported, alpha2 had the issue, I could reproduce:

```
Configuration error: Bootloader not supported on this target. ROM start not found in targets.json.
Configuration error: Bootloader not supported on this target. RAM start not found in targets.json.
```

but not on latest master, means it is fixed and will be in beta1 we release. Can you confirm @ifyall  ?

vmedcy, 2020-04-02:
@0xc0170 : the bootloader error was fixed by https://github.com/ARMmbed/mbed-os/pull/12679 - now arm_pack_manager/index.json defines memory regions for all PSoC 62/63 MPNs (Cypress.PSoC6_DFP CMSIS pack is now included in Keil pack index). That PR introduced another unexpected error for PSoC 6 targets (wrong MBED_RAM_START and MBED_RAM_SIZE linker macros), I am in process of fix preparation/testing.

I am disappointed by non-trivial dependency between CMSIS exporters (uVision, IAR) and bootloader/linker script related code. It is very non-trivial to follow and not well documented.

0xc0170, 2020-04-27:
@vmedcy I recall this was fixed, can it be closed?

ifyall, 2020-04-28:
@0xc0170 , I cannot reproduce this error using b5f00bc67adde82d2e5ffb9e81e4afea8365a1f3. I will close and if this shows back up in the upcoming Beta release, I will re-open.
Thanks,
Ian

closed 2020-04-28 by ifyall