ARMmbed/mbed-os #12538: targets.json: default_lib is no longer supported, but is still used

JojoS62, 2020-02-29:
### Description of defect

A 'mbed compile -m LPC824 -t GCC_ARM' complains about 
```
Could not compile for LPC824: target.default_lib is no longer supported, please use target.c_lib for C library selection.
```

#### Target(s) affected by this defect ?
The setting 'default_lib' is used 37 times in the actual targets.json

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM, gcc 9.2

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15 sha 06b22d91e3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
compile mbed-os-example-blinky with [bare-metal].

0xc0170, 2020-03-06:
@JojoS62 Can you update the target and send PR please?

0xc0170, 2020-03-06:
cc @ARMmbed/mbed-os-core

evedon, 2020-03-06:
`targets.json` for Mbed 2 targets was not modified when we renamed `default_lib` to `c_lib`. 
I believe that the 37 targets are Mbed 2 targets.

JojoS62, 2020-03-06:
Isn't the bare_metal and small lib the way to migrate mbed2 targets to mbed5? By using these options, mbed5 becomes usable for targets with small memory.

evedon, 2020-03-06:
I was only providing explanation for why `default_lib` is still present in targets.json.

Indeed, bare metal and small lib is the way to migrate Mbed 2 targets. In the next release we will provide a migration guide with step by step instructions. 
We welcome PRs to help do the migration.

JojoS62, 2020-03-06:
ok, thanks. 
PR is started.

closed 2020-03-09 by JojoS62