ARMmbed/mbed-os #12539: Using the profile extension lto produces many 'Unknown object name found in GCC map file' messages

JojoS62, 2020-02-29:
### Description of defect

compiling with the new option '--profile mbed-os/tools/profiles/extensions/lto.json' produces a lot of messages like:
```
Unknown object name found in GCC map file: C:\Users\sn\AppData\Local\Temp\blinky.elf.Z2hGQv.ltrans0.ltrans.o
```
This is caused by the memap.py tool that analyses the mapfile.
The lto option generates intermediate files that are stored in a temporary directory, under windows by default in the 'c:\user\xx\AppData\Local\Temp' dir.  The memap tool is parsing the mapfile and there the regex for the objectfiles does not recognize this path.
I have patched the memap.py to ignore the *.ltrans.o files and will provide a PR for this.

#### Target(s) affected by this defect ?
all, toolchain related

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os master sha 06b22d91e3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2, Windows10

#### How is this defect reproduced ? 
compile any program with profile extension lto.json

0xc0170, 2020-03-02:
cc @fkjagodzinski

JojoS62, 2020-03-04:
I have also a  fix for #12541. Should I supply a seperate PR or include the commit here?

0xc0170, 2020-03-06:
> I have also a fix for #12541. Should I supply a seperate PR or include the commit here?

Separate PR will work, thanks!

JojoS62, 2020-03-11:
PR is merged.

closed 2020-03-11 by JojoS62