ARMmbed/mbed-os #12541: memap tool: displayed text+data sizes differ from real size

JojoS62, 2020-02-29:
### Description of defect

By checking the memap tool for https://github.com/ARMmbed/mbed-os/issues/12539
I found that the printed text and data sizes differ from the real sizes

#### Target(s) affected by this defect ?
all? Tested on BLUEPILL_F103C8 output, the difference appears for debug and release build:

```
> Executing task in folder blinky: mbed compile --build ../../BUILD/blinky/Debug --source . --source ../../mbed-os --source ../../custom_targets --profile debug <

[mbed] Working path "d:\Projects\Sn\LocalGit\Mbed5-Projects\apps\blinky" (program)
Building project blinky (BLUEPILL_F103C8, GCC_ARM)
Scan: blinky 
Scan: mbed-os
Scan: custom_targets
Configuration error: Bootloader not supported on this target. ROM start not found in targets.json.
Configuration error: Bootloader not supported on this target. RAM start not found in targets.json.
Link: blinky
Elf2Bin: blinky
| Module           |     .text |   .data |    .bss |
|------------------|-----------|---------|---------|
| [fill]           |    32(+0) |   4(+0) |  17(+0) |
| [lib]\c_nano.a   |  4346(+0) | 100(+0) |  21(+0) |
| [lib]\gcc.a      |   760(+0) |   0(+0) |   0(+0) |
| [lib]\misc       |   180(+0) |   4(+0) |  28(+0) |
| main.o           |   114(+0) |   0(+0) |  28(+0) |
| mbed-os\drivers  |   208(+0) |   0(+0) |   0(+0) |
| mbed-os\hal      |  1780(+0) |   4(+0) |  66(+0) |
| mbed-os\platform |  4586(+0) | 260(+0) | 424(+0) |
| mbed-os\targets  |  7038(+0) |   4(+0) | 368(+0) |
| Subtotals        | 19044(+0) | 376(+0) | 952(+0) |
Total Static RAM memory (data + bss): 1328(+0) bytes
Total Flash memory (text + data): 19420(+0) bytes

Image: ../../BUILD/blinky/Debug\blinky.bin
```

```
d:\Mbed5-Projects\BUILD\blinky\Debug>arm-none-eabi-size blinky.elf
   text    data     bss     dec     hex filename
  21616     392     952   22960    59b0 blinky.elf

d:\Mbed5-Projects\BUILD\blinky\Debug>dir blinky.bin
29.02.2020  15:21            22.008 blinky.bin
               1 Datei(en),         22.008 Bytes
```

#### Toolchain(s) (name and version) displaying this defect ?
gcc 9.2.1

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os master sha 06b22d91e3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
compile any program, verify with 'arm-none-eabi-size output.elf' or size of output.bin

0xc0170, 2020-03-06:
cc @ARMmbed/mbed-os-tools @MarceloSalazar

0xc0170, 2020-04-14:
It was not clear in the fix #12591, what sections were not included in the final image size?

I do not understand the comparison above between elf and bin, can you clarify what are the numbers being reported and what should be?

JojoS62, 2020-04-14:
I have used the gcc size tool as a reference for the correct sizes, this tool reads the sizes from the .elf file.
Yes, text+data should be the .bin size. There the output size from the mbed compile is much lower than the bin size or the report from the arm-none-eabi-size tool.
The fix is not easy because different targets have defined diefferent sections that belong to FLASH or RAM. Idealy, the reported sizes should tell how much flash and (static) ram is used, but the figures are always too low.

JojoS62, 2020-04-14:
yes, thats it. The sizes per module look fine, but the totals do not sum all modules.

ifyall, 2020-05-12:
Also see https://github.com/ARMmbed/mbed-os/issues/10922.

boraozgen, 2021-11-12:
This issue shouldn't be closed. It is just unacceptable to show incorrect information. If memap is not going to get fixed, I propose using the GNU `size` to at least show _correct_ information rather than detailed information about components. If a more detailed view is required, the users can run memap or tools like [mbed-os-linker-report](https://github.com/ARMmbed/mbed-os-linker-report) after the build.

swapnilsayansaha, 2021-11-22:
I am having the same issue. This is not acceptable from Mbed's part. In particular, Mbed Studio shows more resource usage (probably correct) than Mbed CLI. How am I supposed to know whether GCC_ARM (used by Mbed CLI) or ARM compiler (used by Mbed Studio) is correct?

https://forums.mbed.com/t/why-gcc-arm-and-armc6-show-different-flash-and-ram-usage-for-same-program-on-same-hardware/15335/3