ARMmbed/mbed-os #12542: compile profile extensions lto and minimal-printf do not work together

JojoS62, 2020-02-29:
### Description of defect

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

idea--list, 2020-02-29:
@JojoS62 
I know it will not fix your recent findings, but consider reverting back to V5.14.2 if you can. I mean it seems in V5.15.0 something has been changed in the core that introduces some unexpected ""fun"" of different kinds in different targets.

JojoS62, 2020-02-29:
thanks, but the lto and printf-minimal extensions are very interesting new features. Specially for tiny targets. That is really great, I can use mbed5 now even with a LPC824, instead of using mbed2.
This missing linker symbol is not a big issue, I will look also for a solution.

JojoS62, 2020-02-29:
the problem maybe related to this:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=88643

I found also a similiar workaround for other wrappers here:
https://github.com/ARMmbed/mbed-os/blob/06b22d91e3fed9e7d98e8975cfcb54e679218be5/platform/source/mbed_alloc_wrappers.cpp#L351-L358

0xc0170, 2020-03-02:
@JojoS62 Can you create a pull request with this "workaround" to start discussion about this?

JojoS62, 2020-03-02:
this workaround was for another function, but I think the same problem. I could not fix minimal_printf and lto with a workaround yet, but I still can try it.
I have also tried to run this on Linux (WSL with Ubuntu 18.04), but there happened another problem that maybe related to python3 and its another issue.

JojoS62, 2020-03-03:
The example compiles with lto and minimal_printf when I add the -u option in the linker options:`
lto.json:
```
    "GCC_ARM": {
        "common": ["-flto"],
        "ld": ["-u main -u printf -u vsnprintf"]
    }
```
But the code size is the same as with lto profile only, the wrapping seems not work in this case.
Still testing.

evedon, 2020-03-06:
To use the minimal-printf library, you should override the parameter `target.printf_lib` with the value `minimal-printf` as shown here: https://github.com/ARMmbed/mbed-os/tree/master/platform/source/minimal-printf#usage

Usage of minimal-printf extension profile is deprecated and will be removed in the next Mbed OS release.

JojoS62, 2020-03-06:
ok, thanks for the comments. Using the target configuration is much more conveniant and more common with the configuration system. Will there be also a setting for turning on LTO?
There may be the same problem, using the wrappers with LTO.

evedon, 2020-03-10:
> ok, thanks for the comments. Using the target configuration is much more conveniant and more common with the configuration system. Will there be also a setting for turning on LTO?
> There may be the same problem, using the wrappers with LTO.

@jamesbeyond Can you comment?

jamesbeyond, 2020-03-30:
Sorry for the late fixing, but it is coming: https://github.com/ARMmbed/mbed-os/pull/12712

evedon, 2020-04-22:
I believe that this issue has now been fixed.

closed 2020-04-22 by evedon