ARMmbed/mbed-os #12544: Ublox: Setting up APN problem

pilotak, 2020-03-02:
### Description of defect

When connecting to the internet with SARA-G350, setting up APN for first time it takes a lot of time - actually never finish due to a timeout, but when the sequence is sent again after timeout it runs fine. (it looks like mdm is not responding to this command).

I tried to delay this command by adding `rtos::ThisThread::sleep_for(10000);` before https://github.com/ARMmbed/mbed-os/blob/c2c6d251fd6d6d7f5f8d5a920b3ad6ce4df0f8ba/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularContext.cpp#L180 but that didn't help.

I did some test with latest master and compared it with 5.13  - the time to set up everything up, send 4 bytes of data to test server, get them back and disconnect takes:
- 43 sec with OS5.13
- 88 sec with latest OS

If this would be fixed it would take 28s (88s - 60s retry timeout) which would be perfect. Please find below full logs

[os-latest-master.txt](https://github.com/ARMmbed/mbed-os/files/4274984/os-latest-master.txt)
[os-5.13.txt](https://github.com/ARMmbed/mbed-os/files/4274985/os-5.13.txt)

#### Target(s) affected by this defect ?
SARA-G350

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM

#### What version of Mbed-os are you using (tag or sha) ?

#12499 - current master with socket write fix

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli: 1.10.1

#### How is this defect reproduced ? 
```cpp
#include "mbed.h"
#include "UBLOX_AT.h"
#include "CellularContext.h"

#if MBED_CONF_MBED_TRACE_ENABLE
    #include "CellularLog.h"
#endif

EventQueue eQueue(32 * EVENTS_EVENT_SIZE);
CellularContext *mdm;
CellularDevice *mdm_device;

#define MDM_HW_FLOW

class myUblox : public UBLOX_AT {
  public:
    explicit myUblox(FileHandle *fh);

    virtual nsapi_error_t hard_power_on();
    virtual nsapi_error_t hard_power_off();
    virtual nsapi_error_t soft_power_on();
    virtual nsapi_error_t soft_power_off();

  private:
    DigitalOut pwrOn;
    DigitalOut rst;
};

myUblox::myUblox(FileHandle *fh) :
    UBLOX_AT(fh),
    pwrOn(MDM_PWRON_pin, 0),
    rst(MDM_RST_pin, 1) {
}

nsapi_error_t myUblox::hard_power_on() {
    rst.write(0);
    ThisThread::sleep_for(1000);
    return NSAPI_ERROR_OK;
}

nsapi_error_t myUblox::hard_power_off() {
    rst.write(1);

    return NSAPI_ERROR_OK;
}

nsapi_error_t myUblox::soft_power_on() {
    pwrOn.write(1);
    ThisThread::sleep_for(150);
    pwrOn.write(0);
    ThisThread::sleep_for(100);

    return NSAPI_ERROR_OK;
}

nsapi_error_t myUblox::soft_power_off() {
    pwrOn.write(1);
    ThisThread::sleep_for(1000);
    pwrOn.write(0);

    return NSAPI_ERROR_OK;
}

CellularDevice *CellularDevice::get_target_default_instance() {
    static UARTSerial serial(MDM_TX_pin, MDM_RX_pin, 115200);

#if defined(MDM_HW_FLOW)
    serial.set_flow_control(SerialBase::RTSCTS, MDM_RTS_pin, MDM_CTS_pin);
#else
    static DigitalOut rts(MDM_RTS_pin, 0);
#endif

    static myUblox device(&serial);
    return &device;
}

bool serverConnect() {
    debug("Server connect\n");

    nsapi_error_t ret = NSAPI_ERROR_PARAMETER;
    TCPSocket socket;
    const char *echo_string = "TEST";
    char recv_buf[4];

    SocketAddress server;
    ret = mdm->gethostbyname("echo.mbedcloudtesting.com", &server);

    if (ret == NSAPI_ERROR_OK) {
        server.set_port(7);
        ret = socket.open(mdm);

        if (ret == NSAPI_ERROR_OK) {
            debug("open OK\n");

            ret = socket.connect(server);

            if (ret == NSAPI_ERROR_OK || ret == NSAPI_ERROR_IS_CONNECTED) {
                debug("connected\n");
                ret = socket.send((void*) echo_string, strlen(echo_string));

                if (ret >= NSAPI_ERROR_OK) {
                    debug("TCP: Sent %i Bytes\n", ret);
                    nsapi_size_or_error_t n = socket.recv((void*) recv_buf, sizeof(recv_buf));

                    socket.close();

                    if (n > 0) {
                        debug("Received from echo server %i bytes\n", n);
                        return true;

                    } else {
                        debug("Socket recv FAILED: %i\n", n);
                    }

                } else {
                    debug("Socket send FAILED: %i\n", ret);
                }

            } else {
                debug("connect FAILED: %i\n", ret);
            }

        } else {
            debug("open FAILED\n");
        }

    } else {
        printf("Couldn't resolve remote host, code: %d\n", ret);
    }

    return false;
}

void mdmConnect() {
    debug("MDM connect\n");
    nsapi_error_t ret = mdm->connect();

    if (ret == NSAPI_ERROR_OK) {
        debug("Network connected\n");
        serverConnect();
        return;
    }

    debug("Connecting failed\n");
}

bool mdmSetup() {
    debug("Device setup\n");
    mdm = CellularContext::get_default_instance();

    if (mdm != NULL) {
        mdm_device = mdm->get_device();

        if (mdm_device != NULL) {
            mdm_device->hard_power_on();

            uint16_t timeout[8] = {1, 2, 4, 8, 16, 32, 64, 128};
            mdm_device->set_retry_timeout_array(timeout, 8);

            mdm->set_credentials("internet");

            // mdm->attach(mdmCb);
            mdm->set_blocking(true);

            mdmConnect();

            return true;

        } else {
            debug("No interface\n");
        }

    } else {
        debug("No device\n");
    }

    return false;
}

#if MBED_CONF_MBED_TRACE_ENABLE
static Mutex trace_mutex;
static char time_st[50];

static char* trace_time(size_t ss) {
    snprintf(time_st, 49, "[%08llums]", Kernel::get_ms_count());
    return time_st;
}

static void trace_wait() {
    trace_mutex.lock();
}

static void trace_release() {
    trace_mutex.unlock();
}

void trace_init() {
    mbed_trace_init();
    mbed_trace_prefix_function_set(&trace_time);

    mbed_trace_mutex_wait_function_set(trace_wait);
    mbed_trace_mutex_release_function_set(trace_release);

    mbed_cellular_trace::mutex_wait_function_set(trace_wait);
    mbed_cellular_trace::mutex_release_function_set(trace_release);
}
#endif

int main() {
#if MBED_CONF_MBED_TRACE_ENABLE
    trace_init();
#endif

    Thread eQueueThread;

    if (eQueueThread.start(callback(&eQueue, &EventQueue::dispatch_forever)) != osOK) {
        debug("eQueueThread error\n");
    }

    mdmSetup();
    mdm->disconnect();

    while (1) {
        ThisThread::sleep_for(osWaitForever);
    }
}
```

0xc0170, 2020-03-02:
cc @ARMmbed/team-ublox

saddam-dev, 2020-03-04:
Fantastic

pilotak, 2020-03-04:
@saddam-dev what you mean?

fahim-ublox, 2020-03-05:
The delay of 60s came due to error response on at command+UPSD. In initialization phase authentication and PDP context defination look to be culprit this part of code was supposed to be for SARA-R412M. A pull request will be generated shortly.

pilotak, 2020-03-05:
OK, thank you. I will test it ASAP

fahim-ublox, 2020-03-06:
Pull request generated [12589](https://github.com/ARMmbed/mbed-os/pull/12589)

0xc0170, 2020-04-27:
> Pull request generated 12589

Merged, will close this issue as resolved

closed 2020-04-27 by 0xc0170