ARMmbed/mbed-os #12552: ESP8266 set_network static function not working.

star297, 2020-03-02:
### Description of defect

Can't set static parameters on the ESP8266, works for ETHERNET but not WIFI.
Not sure whether the Mbed driver ever has worked tbh.
Connects with no errors, but has default IP.

    #define IP         "192.168.1.180"
    #define GATEWAY    "192.168.1.1"
    #define NETMASK    "255.255.255.0" 

    NetworkInterface* net = NetworkInterface::get_default_instance();  
  
    net->set_network((SocketAddress)IP,(SocketAddress)NETMASK,(SocketAddress)GATEWAY);

    if (!net) {
        printf("No network interface found, select an interface in 'mbed_app.json'\n");
        return NULL;
    }
    nsapi_error_t connect_status = net->connect();
    if (connect_status != NSAPI_ERROR_OK) {
        printf("Failed to connect to network (%d)\n", connect_status);
        return NULL;
    }
    SocketAddress net_addr;   
    net->get_ip_address(&net_addr);
    printf("IP address: %s\n", net_addr.get_ip_address() ? net_addr.get_ip_address() : "None");
    net->get_netmask(&net_addr);
    printf("Netmask:    %s\n", net_addr.get_ip_address() ? net_addr.get_ip_address() : "None");
    net->get_gateway(&net_addr);
    printf("Gateway:    %s\n", net_addr.get_ip_address() ? net_addr.get_ip_address() : "None");    
    printf("MAC:        %s\n", net->get_mac_address());

#### Target(s) affected by this defect ?
Testing on STM32F767ZI, but will affect all that would use ESP8266

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Online

#### What version of Mbed-os are you using (tag or sha) ?

  OS 5.15.1 rev. 7355

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Online

#### How is this defect reproduced ? 

The example will show the problem.

michalpasztamobica, 2020-03-24:
Hi @star297 , thanks for reporting this. It seems that `set_network` is not implemented for ESP8266 driver at all and I suspect it has never been. Can you please check the return value of `net->set_network()` call? I suppose it's NSAPI_ERROR_UNSUPPORTED?

ESP8266 does support static IP address setting in station mode, so this should be doable.

star297, 2020-03-24:
Hi @michalpasztamobica, Yes you could well be correct, I have various projects using both Mbed official ESP8266 driver and my own lightweight library where I do have a set_network() functionality. So may be some confusion there.
Now all of my projects use both Mbed ETHERNET or Mbed WIFI (ESP8266) depending on platform.
Could I would suggest to close this 'BUG' off and add it to your TODO list as and when you can get to it.
This function is very important and would also bring it in line with the ETHERNET functions.

Using Mbed libraries, I run the ESP8266 at 460800 baud on 120Mhz + targets. This has proved extremely reliable over long periods of time, one STM32F767ZI target file server project running continuously for over 4 months. So it's well worth Mbed supporting that ESP8266.

michalpasztamobica, 2020-03-25:
I agree it's worth adding, @star297 . I suggest to leave the issue open. It has already auto-generated an internal ticket to our backlog, so as soon as there's bandwidth, someone will get down to it. 

Also - feel free to contribute the solution via a PR if you have time (Here are some recent notable additions to ESP8266 code, which you could use as an example: https://github.com/ARMmbed/mbed-os/pull/12300, https://github.com/ARMmbed/mbed-os/pull/12234). Please let us know if that is possible so we don't double the effort. I can provide guidance over e-mail if needed. If can get down to this first, you'll see a message in this github issue saying that the internal JIRA status has moved to "in progress".

closed 2020-05-13 by 0xc0170