ARMmbed/mbed-os #12555: USB Host Feature after OS 5.15.0

cyliangtw, 2020-03-03:
### Description of defect

In versions of Mbed OS 5.15.0, mbed could support both of USB-host & USB-device features by  `//mbed-os/features/unsupported/`. 
In current master branch of Mbed OS, removed `//mbed-os/features/unsupported/` and just keep USB-device only in `//mbed-os/drivers/source/usb` .
Why not support USB-host anymore ?

For LTE/NB-IoT module usb interface, USBHost3GModule or USBHostSerial seems useful. Is there any plan for USB-host ?

idea--list, 2020-03-03:
@cyliangtw 
My understanding is that USBDEVICE has been long an EXPERIMENTAL feature of mbed-os core. Until mbed-os V5.10 it was not OFFICIALLY part of the library, though it was available for use.
Then as with V5.10 the mbed team has refactored the USBDEVICE implementation but with breaking changes. This is why the USB functionality got broken for almost all the boards, however since that time it has been fixed for some boards, while for other boards it is still broken. Actually the old implementation became unsuppoerted since mbed V5.10 the that is why you can find the old code under `//mbed-os/features/unsupported/`. Under `//mbed-os/drivers/source/usb` you can find the new and official implementation but as said that is still broken for some boards. 
The very same thing confused me as well earlier, i think it would make more sense to remove the old code even from the unsupported folder to avoid confusing people.

cyliangtw, 2020-03-03:
@idea--list 
Thanks of your comment, I realized Mbed usb feature history and current Mbed usb-device is well.
My inquiry is for some **usb-host feature** framework or implementation, then I could follow, just like as //mbed-os/features/unsupported/USBHost.

0xc0170, 2020-03-06:
Thanks @idea--list for the details.

We did not want to remove unsupported folder as it was used and not all functionality there was implemented for Mbed OS. The upcoming Mbed OS 6 version, will have this cleaned up. One downside is that Host is yet not supported. I'll seek more detailed answer.

@cyliangtw This would be better to discuss on the forum. I'll close this, please redirect this to discussion on the forum.

0xc0170, 2020-03-06:
I closed this, but will find out more and get back.

closed 2020-03-06 by 0xc0170