ARMmbed/mbed-os #12556: Can't install 'cmsis_pack_manager'

ricky-3i, 2020-03-03:
### Description of defect
When trying to generate firmware CA keys and certificates by executing `mbed dm init <params>`, encountered warning about some missing Python modules (namely idna, cmsis_pack_manager, psutil). Then tried installing the missing modules, but encountered error while installing cmsis_pack_manager. [err_msg.txt](https://github.com/ARMmbed/mbed-os/files/4279557/err_msg.txt)

All the above was done using Python 3.7.3 and pip version 19 (I was told that there is a known issue with version 20)

#### Target(s) affected by this defect ?
-

#### Toolchain(s) (name and version) displaying this defect ?
-

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.14.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ?

soleilplanet, 2020-03-03:
@ricky-3i since you are using Linux in your product and just need the DM features from the tools, I would assume upgrading mbed-os and tools wouldn't be an issue for you, would you please try if mbed-os-5.15.2 and mbed-cli 1.10.2 and see if it works better? Thanks!

@madchutney would you please help? Thanks!

ricky-3i, 2020-03-03:
Hi, just updated mbed-cli to 1.10.2. retrying to install cmsis_pack_manager still result in same error. Can you please provide guide (links?) on upgrading the mbed-os? Thank you!

madchutney, 2020-03-03:
The reason the `cmsis-pack-manager` fails to install is probably because the PyPI package does not include a binary suitable for a Raspberry Pi (which appears to be the development platform being used). Can you please retry on a main line Linux distribution such as Ubuntu?

This issue has a similar error to the issue https://github.com/ARMmbed/mbed-cli/issues/949 which occurs on all platforms due to an error in `pip` version `20.0` which tried to compile the package for the host platform even if a binary was available.

ricky-3i, 2020-03-04:
@madchutney Yes currently using Raspberry Pi 3 (Raspbian 10), so should I still use the Pi but using Ubuntu (Any specific version, is Ubuntu MATE okay?) or should I use another Ubuntu from another PC?

madchutney, 2020-03-04:
Hi @ricky-3i, my apologies I should have been clearer. I'm afraid that this package is specified as `manylinux1` which only includes support for `x86_64` and `i686` and will therefore not work on a Raspberry Pi. You'll need to use a PC with either a `x86_64` and `i686` architecture and standard linux distribution as your host development platform to use these tools.

JojoS62, 2020-03-07:
I had also installed mbed-cli on a RPi4, just for testing. I ran into the same trouble and tried to compile the cmsis-pack-manager, but it ended up with installing a lot of rust programming stuff and still producing errors.
Then I removed the cmsis-pack-manager from the requirements.txt and at least I was able to compile mbed programs.
If you are using Win10, you can run 'mbed dm init' in the WSL, that seems to work.

madchutney, 2020-03-09:
Hi @JojoS62, the cmsis-pack-manager is written in Rust and binaries are provided for a number of operating systems and architectures. Unfortunately Raspberry PI is not one of them, so when the package is installed an attempt is made by the Python package manager to compile the tools for the host platform.

madchutney, 2020-03-09:
Closing ticket as this is an unsupported feature.

closed 2020-03-09 by madchutney