ARMmbed/mbed-os #12563: MIMXRT1050_EVK compiler error with nanostack-border-router

artokin, 2020-03-03:
### Description of defect

Compiler error when trying to build nanostack-border-router application using target MIMXRT1050_EVK:

```
Compile [  1.9%]: SimpleSpirit1.cpp
[Error] SimpleSpirit1.cpp@14,30: 'us_ticker_read' was not declared in this scope
[Error] SimpleSpirit1.cpp@14,30: 'us_ticker_read' was not declared in this scope
[Error] SimpleSpirit1.cpp@14,30: 'us_ticker_read' was not declared in this scope
[Error] SimpleSpirit1.cpp@14,30: 'us_ticker_read' was not declared in this scope
[Error] SimpleSpirit1.cpp@14,30: 'us_ticker_read' was not declared in this scope
[Error] SimpleSpirit1.cpp@14,30: 'us_ticker_read' was not declared in this scope
[Error] SimpleSpirit1.cpp@14,30: 'us_ticker_read' was not declared in this scope
[ERROR] ./drivers/stm-spirit1-rf-driver/source/SimpleSpirit1.cpp: In member function 'int SimpleSpirit1::send(const void*, unsigned int, bool)':
./drivers/stm-spirit1-rf-driver/source/SimpleSpirit1.cpp:14:30: error: 'us_ticker_read' was not declared in this scope
             uint32_t start = us_ticker_read(); \
                              ^
./drivers/stm-spirit1-rf-driver/source/SimpleSpirit1.cpp:175:5: note: in expansion of macro 'BUSYWAIT_UNTIL'
     BUSYWAIT_UNTIL(SPIRIT1_STATUS() == SPIRIT1_STATE_RX, STATE_TIMEOUT);

```

#### Target(s) affected by this defect ?
MIMXRT1050_EVK

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM, ARMC6

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15.0, master (#6e6603f97d84e8ef9460c73fb3bc8d5de0cd70f7)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
* git clone git@github.com:ARMmbed/nanostack-border-router.git
* mbed deploy
* mbed compile -t ARM -m MIMXRT1050_EVK

0xc0170, 2020-03-06:
cc @maclobdell  @ARMmbed/team-nxp  Please review

adbridge, 2020-03-11:
@artokin I've updated the template for you (as a good example)

mmahadevan108, 2020-03-11:
I am not clear why this error is seen. This function is defined in the driver as seen below
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/TARGET_MCUXpresso_MCUS/TARGET_IMX/us_ticker.c#L89

and the mbed-os ticker tests are passing.