ARMmbed/mbed-os #12564: CY8CPROTO_062_4343W failed on kvstore Greentea tests

jamesbeyond, 2020-03-03:
### Description of defect

In our testing CI, we can see `CY8CPROTO_062_4343W` target is failing on kvstore Greentea tests:
`features-storage-tests-kvstore-general_tests_phase_1` and `features-storage-tests-kvstore-general_tests_phase_2`

The tests start failing between around 27Feb
@ARMmbed/team-cypress do you have any idea about this

#### Target(s) affected by this defect ?
 CY8CPROTO_062_4343W

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6 / GCC_ARM / IAR

#### What version of Mbed-os are you using (tag or sha) ?
Latest master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli (latest released)

#### How is this defect reproduced ? 
```
mbed test -t GCC_ARM -m CY8CPROTO_062_4343W -n features-storage-tests-kvstore-general_tests_phase*
```

kyle-cypress, 2020-03-03:
I just filed #12565 for the same issue. The failures are caused by #12385 (thanks @VeijoPesonen for bisecting to narrow it down). I don't know why yet.

jamesbeyond, 2020-03-03:
Cool, I close this one as duplication

closed 2020-03-03 by jamesbeyond