ARMmbed/mbed-os #12565: KVStore greentea tests broken on some Cypress targets after #12385

kyle-cypress, 2020-03-03:
### Description of defect
Several storage tests fail on some Cypress targets with latest master. This appears to have been caused by #12385 (the same tests pass with a8fe8310b94a814aeca32c31d700fbab9b6cb6f3)

The failing tests are:
- features-storage-tests-kvstore-general_tests_phase_1
- features-storage-tests-kvstore-general_tests_phase_2

Targets tested and failing:
- CY8CPROTO_062_4343W
- CY8CKIT_062_WIFI_BT

Targets tested and passing:
- CY8CPROTO_063_BLE 
Note: this board does not have external QSPI flash and therefore implements KVStore using internal flash. The other two boards use external QSPI flash for kvstore.

[Test log](https://github.com/ARMmbed/mbed-os/files/4283002/test.log) 

#### Target(s) affected by this defect ?
CY8CPROTO_062_4343W
CY8CKIT_062_WIFI_BT
(Possibly others - I have not exhaustively tested all Cypress targets)

#### Toolchain(s) (name and version) displaying this defect ?
GCC 6.3.1

#### What version of Mbed-os are you using (tag or sha) ?
744889a47280dda706f1fe734803efd187199531

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
1.10.1

#### How is this defect reproduced ? 
`mbed test -t GCC_ARM -m CY8CPROTO_062_4343W -n *general_tests*`

jamesbeyond, 2020-03-03:
Hi @tymoteuszblochmobica,
Any ideas about this issue? this seems linked with https://github.com/ARMmbed/mbed-os/pull/12385

0xc0170, 2020-03-04:
@kyle-cypress Thanks for the report. I can confirm this is now also in nightly

@ARMmbed/mbed-os-storage

tymoteuszblochmobica, 2020-03-05:
Im working on it

tymoteuszblochmobica, 2020-03-09:
PR with fix
https://github.com/ARMmbed/mbed-os/pull/12599

closed 2020-03-11 by ciarmcom