ARMmbed/mbed-os #12566: HardFault in stm32_emac_thread when traces are enabled

artokin, 2020-03-04:
### Description of defect

HardFault detected in nanostack-border-router using DISCO_F769NI with traces enabled:

```
2020-03-03T22:22:45.966Z | D1 <<< DutThread: [INFO][cApp]: Backhaul: EMAC
2020-03-03T22:22:45.966Z | D1 <<< DutThread: [INFO][STE1]: power_up
2020-03-03T22:22:45.966Z | D1 <<< DutThread: [INFO][STE1]: PHY Addr 0 AutoNegotiation 1
2020-03-03T22:22:45.966Z | D1 <<< DutThread: [DBG ][STE1]: MAC Addr 00:80:e1:33:00:32
2020-03-03T22:22:45.967Z | D1 <<< DutThread: [INFO][STE1]: ETH buffers : 4 Rx 4 Tx
2020-03-03T22:22:46.807Z | D1     Dummy-1: Waiting for response... timeout=328
2020-03-03T22:22:47.669Z | D1 <<< DutThread: [INFO][STE1]: PHY ID 0X7C130
2020-03-03T22:22:47.669Z | D1 <<< DutThread: [INFO][STE1]: low_level_init_successful
2020-03-03T22:22:47.808Z | D1     Dummy-1: Waiting for response... timeout=327
2020-03-03T22:22:47.870Z | D1 <<< DutThread: [INFO][STE1]: emac_link_state_cb set to true
2020-03-03T22:22:47.870Z | D1 <<< DutThread: [DBG ][pIP6]: IPV6 interface Base Ready
2020-03-03T22:22:47.870Z | D1 <<< DutThread: [DBG ][rout]: Added route:
2020-03-03T22:22:47.870Z | D1 <<< DutThread: [DBG ][rout]:                    fe80::/64  if:1 src:'Static' id:0 lifetime:infinite
2020-03-03T22:22:47.871Z | D1 <<< DutThread: [DBG ][rout]:      On-link (met 128)
2020-03-03T22:22:47.871Z | D1 <<< DutThread: [DBG ][rout]: Added route:
2020-03-03T22:22:47.871Z | D1 <<< DutThread: [DBG ][rout]:                    ff00::/8   if:1 src:'Static' id:0 lifetime:infinite
2020-03-03T22:22:47.871Z | D1 <<< DutThread: [DBG ][rout]:      On-link (met 192)
2020-03-03T22:22:47.871Z | D1 <<< DutThread: [INFO][addr]: Tentative Address added to IF 1: fe80::280:e1ff:fe33:32
2020-03-03T22:22:47.872Z | D1 <<< DutThread: [DBG ][addr]: LL64 Register OK!
2020-03-03T22:22:48.571Z | D1 <<< DutThread: [DBG ][STE1]: low_level_input len 110
2020-03-03T22:22:48.572Z | D1 <-- DutThread: ++ MbedOS Fault Handler ++
2020-03-03T22:22:48.573Z | D1 <-- DutThread: FaultType: HardFault
2020-03-03T22:22:48.573Z | D1 <-- DutThread: Context:
2020-03-03T22:22:48.573Z | D1 <-- DutThread: R0   : 2000357C
2020-03-03T22:22:48.573Z | D1 <-- DutThread: R1   : 00000001
2020-03-03T22:22:48.573Z | D1 <-- DutThread: R2   : 200182C1
2020-03-03T22:22:48.574Z | D1 <-- DutThread: R3   : 71F4A0DD
2020-03-03T22:22:48.574Z | D1 <-- DutThread: R4   : 2000357C
2020-03-03T22:22:48.574Z | D1 <-- DutThread: R5   : 200035C8
2020-03-03T22:22:48.574Z | D1 <-- DutThread: R6   : 00000000
2020-03-03T22:22:48.575Z | D1 <-- DutThread: R7   : 20003A70
2020-03-03T22:22:48.575Z | D1 <-- DutThread: R8   : 00000000
2020-03-03T22:22:48.575Z | D1 <-- DutThread: R9   : 08004799
2020-03-03T22:22:48.673Z | D1 <-- DutThread: R10  : 00000000
2020-03-03T22:22:48.673Z | D1 <-- DutThread: R11  : 00000000
2020-03-03T22:22:48.674Z | D1 <-- DutThread: R12  : 2000357C
2020-03-03T22:22:48.674Z | D1 <-- DutThread: SP   : 2005F148
2020-03-03T22:22:48.674Z | D1 <-- DutThread: LR   : 0800524D
2020-03-03T22:22:48.674Z | D1 <-- DutThread: PC   : 71F4A0DC
2020-03-03T22:22:48.675Z | D1 <-- DutThread: xPSR : 610F0000
2020-03-03T22:22:48.675Z | D1 <-- DutThread: PSP  : 2005F0E0
2020-03-03T22:22:48.675Z | D1 <-- DutThread: MSP  : 2007FFC0
2020-03-03T22:22:48.675Z | D1 <-- DutThread: CPUID: 411FC270
2020-03-03T22:22:48.676Z | D1 <-- DutThread: HFSR : 40000000
2020-03-03T22:22:48.676Z | D1 <-- DutThread: MMFSR: 00000001
2020-03-03T22:22:48.676Z | D1 <-- DutThread: BFSR : 00000000
2020-03-03T22:22:48.676Z | D1 <-- DutThread: UFSR : 00000000
2020-03-03T22:22:48.676Z | D1 <-- DutThread: DFSR : 0000000B
2020-03-03T22:22:48.677Z | D1 <-- DutThread: AFSR : 00000000
2020-03-03T22:22:48.677Z | D1 <-- DutThread: Mode : Thread
2020-03-03T22:22:48.677Z | D1 <-- DutThread: Priv : Privileged
2020-03-03T22:22:48.677Z | D1 <-- DutThread: Stack: PSP
2020-03-03T22:22:48.677Z | D1 <-- DutThread: -- MbedOS Fault Handler --
2020-03-03T22:22:48.677Z | D1 <-- DutThread: ++ MbedOS Error Info ++
2020-03-03T22:22:48.678Z | D1 <-- DutThread: Error Status: 0x80FF013D Code: 317 Module: 255
2020-03-03T22:22:48.678Z | D1 <-- DutThread: Error Message: Fault exception
2020-03-03T22:22:48.678Z | D1 <-- DutThread: Location: 0x71F4A0DC
2020-03-03T22:22:48.678Z | D1 <-- DutThread: Error Value: 0x20059B34
2020-03-03T22:22:48.679Z | D1 <-- DutThread: Current Thread: stm32_emac_thread Id: 0x20018134 Entry: 0x80211AD StackSize: 0x200 StackMem: 0x2005EFC8 SP: 0x2005F148
2020-03-03T22:22:48.679Z | D1 <-- DutThread: For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=DISCO_F769NI
2020-03-03T22:22:48.679Z | D1 <-- DutThread: -- MbedOS Error Info --

```

#### Target(s) affected by this defect ?
DISCO_F769NI

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
* Build nanostack-border-router application for DISCO_F769NI using S2868A1 radio
* Start the application, the HardFault happens in startup.

0xc0170, 2020-03-04:
This is the cause most probably: https://github.com/ARMmbed/mbed-os/commit/b3583f04cfb63cb87f302fafbda79ff391b0a2e5

cc @ARMmbed/team-st-mcd Please review - it would be good to have this fixed today (fix or revert) to get nightly green again

0xc0170, 2020-03-04:
cc @jeromecoutant

jeromecoutant, 2020-03-04:
> This is the cause most probably: [b3583f0](https://github.com/ARMmbed/mbed-os/commit/b3583f04cfb63cb87f302fafbda79ff391b0a2e5)

Yes, I agree.
I know this issue and opened an issue: #12561 
A workaround is ready in my dev environment, I can push it.

0xc0170, 2020-03-04:
@jeromecoutant Let's fix master asap as it's blocking other teams (testing fails with latest master). We can just disable completely the tracing in the emac file until its fixed. I'll wait for the fix to review

PR #12567 (2020-03-09): STM32 EMAC : increase thread size when mbed-trace is enabled
### Summary of changes 

Discussion started in #12561 

Fix #12566 

In case of mbed trace enable, stm32_emac_thread size has to be increased, in particular for GCC.

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type