ARMmbed/mbed-os #12568: Missing definition of __SXTB16_RORn in cmsis_gcc.h

freddan80, 2020-03-04:
### Description of defect

In the latest cmsis repo, there's a definition of __SXTB16_RORn in cmsis_gcc.h. It is missing in:

https://github.com/ARMmbed/mbed-os/blob/master/cmsis/TARGET_CORTEX_M/cmsis_gcc.h

since it old. Latest CMSIS cmsis_gcc.h:

https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core/Include/cmsis_gcc.h

Do you mind updating that file/folder?

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?
Latest

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

freddan80, 2020-03-04:
> @freddan80 thank you for raising this issue.Please take a look at the following comments:
>
> What target(s) are you using?
> What toolchain(s) are you using?
> Have you correctly specified which version [...]

All targets with DSP extensions
GCC
mbed-cli 1.10.0

Instructions on how to reproduce can be found:
https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/kernels/cmsis-nn

adbridge, 2020-03-05:
@freddan80 please update the template with the missing information or add n/a or None to fields that are not applicable so that our automated checker can resolve them

freddan80, 2020-03-05:
Done

freddan80, 2020-03-05:
I use the version downloaded by mbed cli. There's no other version specified anywhere.

0xc0170, 2020-03-05:
if you get to mbed-os, you can git log to find out the sha of Mbed OS.

Anyway, this file should be updated then.

@ARMmbed/mbed-os-core Would it make sense to update CMSIS files for 6.0? I recall there was a ticket for it or at least was touched few weeks back.

evedon, 2020-03-05:
There is a ticket in the backlog. It needs to be prioritised by our product manager.

adbridge, 2020-03-11:
>> What version of Mbed-os are you using (tag or sha) ?
commit e642a7d (HEAD, tag: mbed-os-5.15.1, tag: latest, origin/mbed-os-5.15)

@freddan80 please provide just the actual sha or a valid release tag so that our automation can validate the field correctly.
E.g
#e642a7d  or mbed-os-5.x.y

freddan80, 2020-03-11:
Done!

freddan80, 2020-03-20:
To expand the context a bit on this issue. There was a similar issue some time ago (https://github.com/ARMmbed/mbed-os/pull/12040), which has now been fixed. Is it possible to solve this issue in a more general fashion, but not having files duplicated from CMSIS in the mbed repo? To avoid more similar issues in the future...

Right now, this is causing some confusion in the community when using CMSIS and Mbed together.

0xc0170, 2020-04-09:
> To expand the context a bit on this issue. There was a similar issue some time ago (#12040), which has now been fixed. Is it possible to solve this issue in a more general fashion, but not having files duplicated from CMSIS in the mbed repo? To avoid more similar issues in the future...

Ideally yes, we would use some form of package manager (cmsis-pack manager was going that direction). In the current form, the build system and this repo requires to have everything present.
A user should be aware CMSIS clone is part of the Mbed OS (or any other modules like lwip, etc).

> There is a ticket in the backlog. It needs to be prioritised by our product manager.

@freddan80 If this is critical for you, you can provide PR updating the core files.

freddan80, 2020-04-14:
Not critical. Do you have a rough ETA for the fix?

0xc0170, 2020-04-14:
There is not ETA unfortunately. I would just patch the file (all toolchains or if only valid for gcc, only that file).

I can attempt to update to  see if its trivial.

0xc0170, 2020-04-14:
I had one conflict that I hope I resolved correctly, here's a branch https://github.com/ARMmbed/mbed-os/compare/master...0xc0170:feature_CMSIS_5_b5f0603d6?expand=1

I dont understand why some files are "new" there like rtx_mutex that already is in our codebase (looks like the importer removed-add these files?). More work needed there.

0xc0170, 2020-04-14:
> I had one conflict that I hope I resolved correctly, here's a branch https://github.com/ARMmbed/mbed-os/compare/master...0xc0170:feature_CMSIS_5_b5f0603d6?expand=1

@kjbracey-arm you run the exporter previously, would you know what did happen on my tryout here as some files seems to be like "new", rtx_system.c or rtx_mutex.c.

kjbracey-arm, 2020-04-14:
Our in-tree location has moved down to `rtos/source`. Has the exporter not been updated for that?

0xc0170, 2020-04-14:
That would be it, checking

0xc0170, 2020-04-14:
Due to the rename, cherry-pick fails for rtx files (they were moved under source directory), this needs a manual resolution. I push an update to my branch to include only cmsis changes and RTOS changes needs a manual resolution now.

0xc0170, 2020-05-11:
PR created to update: https://github.com/ARMmbed/mbed-os/pull/12949

closed 2020-05-14 by 0xc0170

EmbeddedSysDesign, 2023-01-12:
Hello, I am facing the same issue during the compilation of a project Hello Edge KWS using mbedOS. Please share the same files that you used in example. See my error details below;

Compile [ 98.4%]: arm_max_pool_s8.c
Compile [ 98.6%]: arm_svdf_s8.c
Compile [ 98.7%]: arm_softmax_s8.c
Compile [ 98.9%]: arm_softmax_s8_s16.c
Compile [ 99.0%]: arm_softmax_s16.c
Compile [ 99.2%]: arm_nn_softmax_common_s8.c
Compile [ 99.3%]: arm_softmax_u8.c
Compile [ 99.4%]: plot_utils.cpp
[Error] arm_nnsupportfunctions.h@681,54: '__SXTB16_RORn' was not declared in this scope
[ERROR] In file included from ../Source/local_NN/local_NN.h:1:0,
from ../Source/NN/DS_CNN/ds_cnn.h:24,
from ../Source/KWS/KWS_DS_CNN/kws_ds_cnn.h:23,
from ../Source/KWS_F746NG/kws_f746ng.h:23,
from ../Examples/realtime_test\plot_utils.h:19,
from ../Examples/realtime_test\plot_utils.cpp:19:
../CMSIS_5/CMSIS/NN/Include/arm_nnsupportfunctions.h: In function 'const int8_t* read_and_pad(const int8_t*, int32_t*, int32_t*)':
../CMSIS_5/CMSIS/NN/Include/arm_nnsupportfunctions.h:681:54: error: '__SXTB16_RORn' was not declared in this scope
int32_t inAbuf1 = __SXTB16_RORn((uint32_t)inA, 8U);
^

[mbed] ERROR: "C:\Python27\python.exe" returned error.
Code: 1
Path: "C:\prj_KWS\Deployment\ML-KWS-for-MCU\Deployment\kws_realtime_test"
Command: "C:\Python27\python.exe -u C:\prj_KWS\Deployment\ML-KWS-for-MCU\Deployment\kws_realtime_test\tools\make.py -t GCC_ARM -m DISCO_F746NG --profile ../release_O3.json --source . --source ../Source --source ../Examples/realtime_test --source ../CMSIS_5/CMSIS/NN/Include --source ../CMSIS_5/CMSIS/NN/Source --source ../CMSIS_5/CMSIS/DSP/Include --source ../CMSIS_5/CMSIS/DSP/PrivateInclude --source ../CMSIS_5/CMSIS/DSP/Source --source ../CMSIS_5/CMSIS/Core/Include --build .\BUILD\DISCO_F746NG\GCC_ARM-RELEASE_O3 -j 8"
Tip: You could retry the last command with "-v" flag for verbose output

PR #12949 (2020-05-14): CMSIS: update to CMSIS 5.7.0

### Summary of changes 

Fixes #12568

There are couple of fixes here:

- importer be compatible with python 3
- importer sha for RTOS after move (sha updated , after I manually resolved the conflicts)

#### Impact of changes 

#### Migration actions required