ARMmbed/mbed-os #12573: Cannot config ublox odin w2 wifi

se7ensong, 2020-03-04:
### Description of defect

Cannot use set_config() API provided by the Ublox driver in the OdinWiFiInterface.cpp to configure wifi parameters such as scan listen interval, remain on channel, etc.

#### Target(s) affected by this defect ?

MODULE_UBLOX_ODIN_W2 family

#### Toolchain(s) (name and version) displaying this defect ?

IAR 8.42.1

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.14.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

IAR 8.42.1

#### How is this defect reproduced ? 

In any example using , add the following code 

```cpp
OdinWiFiInterface *network;
network  = (OdinWiFiInterface *)(WiFiInterface::get_default_instance());
target_config_params_e cfg = ODIN_CFG_SET_SCAN_LISTEN_INTERVAL;
network->set_config(&cfg, 150);
```

se7ensong, 2020-03-05:
Not sure what should I specify here. The tag shown on the github is indeed mbed-os-5.14.2

adbridge, 2020-03-05:
> 
> 
> Not sure what should I specify here. The tag shown on the github is indeed mbed-os-5.14.2

@se7ensong  I'll investigate, that tag is indeed valid and should have been correctly identified by our automation

JanneKiiskila, 2020-04-07:
@ARMmbed/team-ublox  - probably up your alley?

qasim-ublox, 2020-04-08:
@se7ensong we've started looking into the issue. 
I have run the application and I've seen that application crashes when `network->set_config(&cfg, 150);` is called. I want to know that, if application is crashing in your case as well or you've encountered a different output? Just to make sure if we are correctly replicating it.. It would be helpful for us if you can provide the application/error logs..
Thanks

se7ensong, 2020-04-08:
@qasim-ublox , yes, that crashes but you should be able to fix this by fixing the fall through here in the switch cases: https://github.com/ARMmbed/mbed-os/blob/4bc491ac30cf58f763c9060ac6fb310d844de890/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/ublox-odin-w2-drivers/OdinWiFiInterface.cpp#L1999

Then, it should go through without issue but it has no effect I think

qasim-ublox, 2020-04-08:
@se7ensong yes you are right about the fix. We've already applied the above fix but it still crashes. While debugging the driver I found that, this crash is coming from driver. We are further investigating it and will provide the fix soon.

qasim-ublox, 2020-09-09:
This issue is fixed in the latest release i.e. v3.7.2. Can we close this issue now?

closed 2020-09-10 by 0xc0170