ARMmbed/mbed-os #12574: Issue with SFDP parsing of the Sector Map Parameter Table

miteshdedhia7, 2020-03-04:
### Description of defect
The QSPIFBlockDevice and potentially the SPIFBlockDevice implementation has the following issue,
Spec Link: JESD216

**BUG/ENHANCEMENT:** The _sfdp_parse_sector_map_table() function currently only supports a single map descriptor (no commands) that may not be valid/generic implementation that will work for all flash devices. The spec specifies that there can be variable number of configuration detection commands, configurations and regions.
Spec Reference: JESD216 pg 53 section 6.5

#### Target(s) affected by this defect ?
CYW9P62S1_43012EVB_01 (Potentially other targets)

#### Toolchain(s) (name and version) displaying this defect ?
Toolchain: All toolchains

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

miteshdedhia7, 2020-03-04:
Split from #11722

adbridge, 2020-03-05:
@miteshdedhia7 Thanks for providing the version for mbed-os, unfortunately that version just specifies that master is being used but doesn't narrow down a specific point. If you are indeed using master could you please instead provide a sha ?

miteshdedhia7, 2020-03-06:
Hi Anna, Ive updated the comment with the commit. This was originally part of #11722 but there was a request to pull these out into separate issues hence submitting this issues.

adbridge, 2020-03-09:
> 
> 
> Hi Anna, Ive updated the comment with the sha that the issue was identified at. This was originally part of #11722 but there was a request to pull these out into separate issues hence submitting this issues.

@miteshdedhia7 It doesn't look like you added the sha to the correct field: 
What version of Mbed-os are you using (tag or sha) ?
mbed-os-99.99.99

Thanks

miteshdedhia7, 2020-03-09:
> @miteshdedhia7 It doesn't look like you added the sha to the correct field:
> What version of Mbed-os are you using (tag or sha) ?
> mbed-os-99.99.99
>
> Thanks

@adbridge I tried updating just the sha but I still keep getting emails that it was not able to parse the sha hence moved it out into the description section and kept the tag/sha field as master. May be the format wasn't correct? Ive updated it again with just the SHA hopefully this works.

adbridge, 2020-03-10:
@miteshdedhia7 Thanks, whatever you did has fixed it :)

amartsan, 2020-12-02:
Hi,

CYW9P62S1-43012EVB-01 target hardware is using serial flash part number S25FS512S.
https://www.cypress.com/file/216376/download

If I check data sheet for this part, the configuration detect commands are

CR3NV[3] - Bit 3 of CR3NV - Default value 0
CR1NV[2] - Bit 2 of CR1NV - Default value 0
CR3NV[1] - Bit 1 of CR3NV - Default value 0

Since default state of these register values are 0 and the constructed configuration index will also be 0.
However, configuration ID's available in SFDP table are only 01, 03, 05 for this serial flash part and hence SMIF driver cannot find a matching configuration index to populate region information.

Regards,
Sandeep

0xc0170, 2021-01-26:
Thanks @amartsan for looking at this issue.

> However, configuration ID's available in SFDP table are only 01, 03, 05 for this serial flash part and hence SMIF driver cannot find a matching configuration index to populate region information.

Is there anything to fix here or what is the action to resolve this issue?

PR #14989 (2021-09-13): SFDP: Add support for multiple configurations and sector maps

### Summary of changes 

Fixes #12574

This PR adds to following changes, to support CYW9P62S1_43012EVB_01's QSPI flash (S25FS512S):

* A Sector Map Parameter Table contains a sequence of the following descriptors:
  * (Optional) configuration detection command descriptors, one for each command to run to determine the current configuration. This exists only if the flash layout is configurable.
  * Sector map descriptors, one map for each possible configuration. On a flash device with a non-configurable layout, there is only one suc [... cut]
closed 2021-09-13 by Patater