ARMmbed/mbed-os #12577: STM32 EMAC: Ethernet driver initialisation fails in DISCO_F769NI

artokin, 2020-03-05:
### Description of defect

STM ethernet driver initialisation fails in DISCO_F769NI and therefore the driver can't be used.

The error happens when fix https://github.com/ARMmbed/mbed-os/pull/12567 is placed on top of Mbed OS master. The driver was working before PR https://github.com/ARMmbed/mbed-os/pull/12464 was merged.

The error log "HAL_ETH_Init issue" inficates the problem in the trace output:
> [DBG ][core]: Allocate Root Tasklet
> [DBG ][6lo ]: P.Init
> [DBG ][core]: NS Root task Init
> [DBG ][sck ]: Socket Tasklet Generated
> [DBG ][mlme]: SW-MAC driver support rf extension 250000 symbol/seconds  4 us symbol time length
> [INFO][brro]: NET_IPV6_BOOTSTRAP_AUTONOMOUS
> [INFO][app ]: Using EMAC backhaul driver...
> [INFO][STE1]: power_up
> [INFO][STE1]: PHY Addr 0 AutoNegotiation 1
> [DBG ][STE1]: MAC Addr 00:80:e1:2d:00:3e
> [INFO][STE1]: ETH buffers : 4 Rx 4 Tx
> **[ERR ][STE1]: HAL_ETH_Init issue**
> **[ERR ][app ]: Backhaul driver init failed, retval = -3012**
> [DBG ][brro]: Create Mesh Interface

#### Target(s) affected by this defect ?
DISCO_F769NI 

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM, ARM, IAR

#### What version of Mbed-os are you using (tag or sha) ?
master with https://github.com/ARMmbed/mbed-os/pull/12567

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
* Clone git@github.com:ARMmbed/nanostack-border-router.git
* Change Mbed OS to master
* Cherry-pick PR  https://github.com/ARMmbed/mbed-os/pull/12567 on top of Mbed OS
* Build nanostack-border-router application for DISCO_F769NI using S2868A1 radio using command: `mbed compile -t GCC_ARM -m DISCO_F769NI -j 0 --app-config ./configs/Wisun_Stm_s2lp_RF.json --profile debug`
* Start tracing the application
* Observe the trace output, the log indicates error in the EMAC driver initialization

jeromecoutant, 2020-03-05:
Hi
I can reproduce the issue only when no ethernet cable is plugged...
No issue when there is one cable.

jeromecoutant, 2020-03-05:
More details when cable is not plugged.

> [ERR ][STE1]: HAL_ETH_Init issue

TIMEOUT comes after waiting linked status.

jeromecoutant, 2020-03-05:
Could you try to add in your json config:
  "stm32-emac.eth-phy-AutoNegotiation" : "ETH_AUTONEGOTIATION_DISABLE"
?

adbridge, 2020-03-09:
@artokin could you please specify either just the sha or a valid release tag for the mbed-os version. The automation cannot parse something like : 
'master plus #234455'
Thanks

artokin, 2020-03-09:
@adbridge sure, it is now pointing to Mbed OS master.

jeromecoutant, 2020-03-11:
> Could you try to add in your json config:
> "stm32-emac.eth-phy-AutoNegotiation" : "ETH_AUTONEGOTIATION_DISABLE"

Did you try this ?
Should we close the issue?

artokin, 2020-03-11:
Hi @jeromecoutant , I'm sorry that I haven't got time to try it yet.

I would like to see the driver working like it worked before (the driver initialisation was OK even if there was no cable connected. And when the cable was connected later on the application was good to continue its operation).
With existing behaviour the application should make some kind of retry logic to handle special case when application was started without Ethernet cable.

JojoS62, 2020-03-11:
I have looked at the sources and see a difference in the behaviour between V2 (for STM32H7 targets) and V1 (all others):
in V2 it is implented better, after the link is up, the phy speed and duplexmode are set. This is done every time the link was down and is up again.
In V1, the phy init incl. setting speed / mode is done only at ethernet init. A link up and down is also detected, but speed/mode are not changed again. When the cable is not connected at startup, then the init fails and eth will not come up when the cable is plugged in now.
I have tried also autoneg=false, but this works only when the settings at the switch are the same. With switch in auto mode, I got no connection.
So a clean solution seems to be some more work. I can test to use the code for the LAN8742 phy from V2, this looks better. When there was so much refactoring, why wasn't the V2 version used for all phy chips? It would be even cleaner to move the phy code into an external hw component that could be used by any target. For ST boards this solution is ok of course, but for custom targets, this thight coupling is unfortunately.

jeromecoutant, 2020-03-13:
> in V2 it is implemented better

I think this was the goal of v2 :-)
For a short term patch, I think I am going to revert:
https://github.com/ARMmbed/mbed-os/pull/12464/commits/c3653c681346cb1300b6157bc4b69f9ecbbfd2ad

jeromecoutant, 2020-03-13:
Does this patch
https://github.com/jeromecoutant/mbed/commit/488c9db81e891dd8f640b98b14ecca251c28c5d8
is OK for you?

0xc0170, 2020-03-24:
PR sent with fix ^^ , please review

PR #12686 (2020-03-25): STM32 EMAC : remove TIMEOUT issue when cable is not plugged yet
### Summary of changes 

This fixes #12577 

ST Ethernet river implementation is not very well implemented...
Init procedure needs Ethernet cable to be plugged in order to return OK!
Workaround is then to not return this error case to higher layer in this case.

@LMESTM 
@JojoS62 
@artokin 

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2020-03-25 by 0xc0170