ARMmbed/mbed-os #12582: BLE stops advertising when WiFi is connected on EVK-ODIN-W2

se7ensong, 2020-03-05:
### Description of defect

When using BLE with WiFi on EVK-ODIN-W2, the BLE stops advertising as soon as `WiFiInterface::get_default_instance();` is called. The BLE doesn't readvertise when WiFI is disconnected. The expected behaviour is BLE stays advertising regardless of WiFi behaviour. 

#### Target(s) affected by this defect ?
MODULE_UBLOX_ODIN_W2 family

#### Toolchain(s) (name and version) displaying this defect ?
mbed online compiler

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.14.2
mbed-os-5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed online compiler

#### How is this defect reproduced ? 

Examples provided here https://os.mbed.com/users/se7ensong/code/mbed-os-example-wifi-ble/

se7ensong, 2020-03-05:
Just want to point out one more thing that if wifi is started before BLE, then it will crash the app.

0xc0170, 2020-03-05:
cc @ARMmbed/team-ublox

aqib-ublox, 2020-03-13:
we have started looking into issue, will update reproducibility and how fix to be provided

se7ensong, 2020-03-31:
@aqib-ublox , is there any update? Thank you.

aqib-ublox, 2020-03-31:
@se7ensong yes we did proceed with some sort of changes,
We have modified application and updated libs, we succeed with enabling Wifi first and then ble advertisement?
Would it fulfill your requirement?

se7ensong, 2020-03-31:
Thank you @aqib-ublox . That is great news! For my application, I would require BLE UART and WIFI to run in parallel regardless which one starts first. Do you think this is something ublox team can test before releasing the code? Or I can help testing. 

One last thing is this still CORDIO stack or UBLOX stack?

aqib-ublox, 2020-03-31:
@se7ensong it would be CORDIO stack no doubt, we would share our internal repo and branch first for testing with application. Application we tested is BLE heart rate service profile with Wifi station, so you can modify that for BLE SPS service. Thanks

aqib-ublox, 2020-04-02:
@se7ensong  sorry for delayed response, can u find https://github.com/u-blox/mbed-os/tree/bt_wifi_demo for testing with GCC and ARM?

aqib-ublox, 2020-04-02:
please see application modified as well [here](https://github.com/aqib-ublox/ExampleCode/tree/master/ble_wifi_example)

se7ensong, 2020-04-04:
Hi @aqib-ublox , I have tested the fix and it works for GCC (2018 q2) and ARMC6. **Can you please also release one for IAR?** The example provided is for easier reproducibility.

qasim-ublox, 2020-04-06:
Hi @se7ensong, I have provided release for IAR as well in the branch already shared. Please have a look and let me know if you face any issue. 

> @se7ensong sorry for delayed response, can u find https://github.com/u-blox/mbed-os/tree/bt_wifi_demo for testing with GCC and ARM?

se7ensong, 2020-04-06:
Thanks @qasim-ublox . Considering master branch will not be supported anymore from the reply here https://github.com/ARMmbed/mbed-os/issues/12720#issuecomment-609664255   I have retested all three compilers with the archive files on 5.15.1 and all of them works. 

Thank you again for your hardwork.

MarceloSalazar, 2020-04-08:
@qasim-ublox thanks for the fixes and @se7ensong for testing.
We're closing this issue as it's considered resolved.

closed 2020-04-08 by MarceloSalazar