ARMmbed/mbed-os #12584: Baremetal / IAR: mbed_sdk_init is called 2 times

jeromecoutant, 2020-03-05:
### Description of defect

In case of IAR compilation and bare metal profile, mbed_sdk_init is called 2 times:

platform/source/TARGET_CORTEX_M/TOOLCHAIN_IAR/cmain.S:        BL      mbed_sdk_init
platform/source/mbed_sdk_boot.c:    mbed_sdk_init();

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-03-05:
cc @ARMmbed/mbed-os-core

adbridge, 2020-03-11:
@jeromecoutant Hi Jerome , for future reference 'master branch' is not a valid mbed-os version that can be tracked properly. Could you please provide an actual sha or release tag ? Thank you.

evedon, 2020-12-17:
This issue is on our backlog, but given our current priorities we can make no
commitment to address this in the near term.

As an open source project, we welcome fixes to Mbed OS, so we have opened this
issue up for community contribution.

If this is an issue you think you could address yourselves, then we'd welcome the
contribution. Please see our guidelines here -
https://os.mbed.com/docs/mbed-os/latest/contributing/workflow.html#contributions .

closed 2021-10-20 by ciarmcom