ARMmbed/mbed-os #12586: PyOCD deploy failed, deploying using mass storage device...

Hammie217, 2020-03-05:
### Description of defect
After compilation of my c++ code after trying to upload to the device I get the error `PyOCD deploy failed, deploying using mass storage device...`
It started happening out of nowhere and I've tried different boards and windows resorting to deleting sections of code to no avail.

The code can be found here:
https://github.com/Hammie217/NXPCupGroupProject/blob/master/main.cpp

#### Target(s) affected by this defect ?
KL25z
K64f

#### Toolchain(s) (name and version) displaying this defect ?
MBED Desktop, PC and Mac

#### What version of Mbed-os are you using (tag or sha) ?

5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
MBED Desktop, PC and Mac

#### How is this defect reproduced ? 
Trying to run the program on windows or mac.

0xc0170, 2020-03-06:
>After compilation of my c++ code after trying to upload to the device I get the error PyOCD deploy failed, deploying using mass storage device...

How are you uploading it to the device? I don't recall seeing before "PyOCD deploy failed" so wondering where this error comes from - please add more details that should help us to reproduce.

Is this related to the example above (I don't find there anything specific to flash or something that could break the board), or blinky as well?

Hammie217, 2020-03-06:
Hi 0xc0170,

I'm uploading to the device directly over USB, I've tried on both macOS and Windows and get the same error.  If you'd like to reproduce I'm sure compiling the linked code onto a similar NXP board will generate the same result.

When I run Blinky it seems to compile, upload and run fine.

Thanks,

adbridge, 2020-03-11:
>> What version of Mbed-os are you using (tag or sha) ?
5.15.1

@Hammie217 For future reference '5.15.1' is not a valid sha or release tag and cannot thus be interpreted by our automation. Please provide the actual tag or a valid sha.
E.g
#absd1234 or mbed-os-x.y.z

Thank you

adbridge, 2020-03-11:
Also:

>>Toolchain(s) (name and version) displaying this defect ?
MBED Desktop, PC and Mac

This is not a toolchain. We would expect something like 'ARMCC' , 'GCC' or 'IAR'

>> What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
MBED Desktop, PC and Mac

This is not a tool set. How are you compiling? If you are using mbed-cli to compile then that should be added in this field. 

Please update the template header with the correct information so that we can try and replicate the problem.
Thanks.

Hammie217, 2020-03-12:
Apologies find the correct formats below.

Version:
mbed-os-5.15.1

Toolchain:
ARMC6

Tool versions:
Mbed Studio (PC and Mac)

Thanks,

nevenaaak, 2020-04-10:
Hello @Hammie217 , I am experiencing the same problem. I was wondering if you have found any solution.

Thanks

Hammie217, 2020-04-10:
Hey @nevenaaak , unfortunately not. I had others have the same issue and again unable to fix.

Apologies,

yvanjo, 2020-05-04:
hi  @Hammie217  , i also have the same problem right now. Have you fix it please?

Neuromod, 2020-05-04:
Same problem here, Cypress PSOC 6 WiFi-BT Pioneer Kit, ARMC 6.13, Mbed Studio 0.91,  Mbed 5.15.0

ParamDeshpande, 2020-07-27:
Same issue, using Nucleo F411RE , mbed studio  1.1.0 , mbed-os 6.2.0

vishnumaiea, 2020-08-04:
Same issue. pyOCD is failing to launch due to which uploading falls back to mass storage method. Debugging is also not working.

[image]

Here's an excerpt from the Mbed Studio log file,

`[2020-08-04T12:21:06.547] [ERROR] MbedStudio - Request deploy failed with error: PyOCD failed to flash the device, please check the logs for more information [ 'Error: PyOCD failed to flash the device, please check the logs for more information\n    at PyocdDeployer.<anonymous> (C:\\Program Files\\Mbed Studio\\resources\\app\\node_modules\\@mbed\\mbs-deploy\\lib\\backend-electron\\deployers\\pyocd-deployer.js:168:31)\n    at step (C:\\Program Files\\Mbed Studio\\resources\\app\\node_modules\\@mbed\\mbs-deploy\\lib\\backend-electron\\deployers\\pyocd-deployer.js:48:23)\n    at Object.throw (C:\\Program Files\\Mbed Studio\\resources\\app\\node_modules\\@mbed\\mbs-deploy\\lib\\backend-electron\\deployers\\pyocd-deployer.js:29:53)\n    at rejected (C:\\Program Files\\Mbed Studio\\resources\\app\\node_modules\\@mbed\\mbs-deploy\\lib\\backend-electron\\deployers\\pyocd-deployer.js:21:65)' ]`

thegecko, 2020-08-07:
Your logs show you are running J-Link firmware on your device which Mbed Studio doesn't support. I recommend you use the DAPLink firmware for your target hardware:

https://armmbed.github.io/DAPLink/

adbridge, 2021-06-07:
Solution to this was provided by @thegecko so closing this.

closed 2021-06-07 by adbridge