ARMmbed/mbed-os #12587: Cellular: How to reconnecting cellular network

matsujirushi, 2020-03-06:
### Description of defect

My environment is not possible to connect to cellular network.
If an error occurs in connect (), calling disconnect () and connect () will freeze.
How is it correct to write code?

**Code snippet:**
```cpp
ctx = (AT_CellularContext*)CellularContext::get_default_instance();
if (ctx == NULL) abort();

while (true)
{
	nsapi_error_t ret;

        ...

	print_function("connect()\n");
	ctx->set_default_parameters();
	const uint16_t timeoutArray[]{ 60 };
	ctx->get_device()->set_retry_timeout_array(timeoutArray, extent<decltype(timeoutArray)>::value);
	if ((ret = ctx->connect()) != NSAPI_ERROR_OK)
	{
		print_function("ret = %d\n", (int)ret);

		print_function("disconnect()\n");
		ret = ctx->disconnect();
		print_function("ret = %d\n", (int)ret);
		continue;
	}

        ...
}
```
[log.txt](https://github.com/ARMmbed/mbed-os/files/4296576/log.txt)

See also:
[Network connectivity states](https://os.mbed.com/docs/mbed-os/v5.15/apis/network-interfaces.html)

#### Target(s) affected by this defect ?

QUECTEL/EC2X ([here](https://github.com/ARMmbed/mbed-os/tree/master/features/cellular/framework/targets/QUECTEL/EC2X))

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli with GCC_ARM

adbridge, 2020-03-09:
@matsujirushi It looks like you have removed some of the template sections and not provided a valid mbed-os version (ie mbed-os-5.15 is a branch not a version :) ). Could you please update the issue header accordingly?

matsujirushi, 2020-03-10:
Hi,

> not provided a valid mbed-os version

mbed-os branch is **mbed-os-5.15.0**.

> What toolchain(s) are you using?

GNU Arm Embedded Toolchain. (Version **9-2019-q4-major**)

> How can we reproduce your issue?

Try connecting in a location where cellular communication is not possible.
I suspended SIM with the function of Soracom SIM. ([here](https://developers.soracom.io/en/docs/air/sim-status/))

AriParkkila, 2020-03-13:
@matsujirushi Can you try without calling `set_retry_timeout_array()`, so that CellularStateMachine tries a few times before giving up.

kivaisan, 2020-03-19:
@matsujirushi Could you please take AT traces of your problem?

0xc0170, 2020-03-25:
> @matsujirushi Can you try without calling set_retry_timeout_array(), so that CellularStateMachine tries a few times before giving up.

As this was answered, I'll close this. Reopen with an update

closed 2020-03-25 by 0xc0170