ARMmbed/mbed-os #12592: How to increase stack size when using IAR toolchain

mmahadevan108, 2020-03-06:
### Description of defect
We have the below code inside the IAR linker script
/* Sizes */
if (!isdefinedsymbol(MBED_BOOT_STACK_SIZE)) {
    define symbol MBED_BOOT_STACK_SIZE = 0x400;
}

define symbol __stack_size__=MBED_BOOT_STACK_SIZE;

I would like to increase the default stack size. However MBED_BOOT_STACK_SIZE is defined by the tools to be 0x400, how can I increase this value?

#### Target(s) affected by this defect ?
MIMXRT1050

#### Toolchain(s) (name and version) displaying this defect ?
IAR

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli with AIR

#### How is this defect reproduced ? 
This is a enquiry on how to increase STACK size when using IAR

mmahadevan108, 2020-03-06:
mbed-os-5.15 branch

0xc0170, 2020-03-09:
This is a question for a forum. I'll close this as a question.

To answer, `MBED_BOOT_STACK_SIZE` is an application specific configuration, set this in an application.

AGlass0fMilk, 2020-03-09:
As 0xc0170 said, you can configure this parameter through [Mbed's configuration system](https://os.mbed.com/docs/mbed-os/v5.15/reference/configuration.html). 

Specifically, I went into a clone of mbed-os and did the following:

`grep -nrw . -e "MBED_BOOT_STACK_SIZE"` and found a reference to the target configuration parameter `target.boot-stack-size` here:

https://github.com/ARMmbed/mbed-os/blob/master/docs/design-documents/platform/memory-model/ram_memory_model.md#detailed-design

To adjust other thread stack sizes for the RTOS, you can find the configurable parameters in this .json file:

https://github.com/ARMmbed/mbed-os/blob/e71ab0d0f020ceba6b3eaea153f4916acdd9284f/rtos/source/TARGET_CORTEX/mbed_lib.json#L1-L29

@0xc0170 Can you close this question?

closed 2020-03-09 by 0xc0170