ARMmbed/mbed-os #12602: BLE: It is no longer possible to send out data Fast!

CodingGhost, 2020-03-09:
Since the update to cordio, it seems impossible to send out a fast datastream.
we used to do it like the NORDIC devs recommended, by transmitting 20 bytes, as fast as we can, and repeat it, until the softdevice wont give error no 5 (busy). This was easy and very fast.
Cordio never givey me error 5, it always is no error, but if I send too fast, data gets lost.
another way would be to register the onDataSent event, but strangely this does not give back any information about the handle that the event is referring to, so its useless if you have more than 1 device connected.
Im not sure if these are bugs, but either way, its very Important!!
How can we do this with cordio?

#### Target(s) affected by this defect ?

NRF52
#### Toolchain(s) (name and version) displaying this defect ?
PlatformIO 4.2.1

#### What version of Mbed-os are you using (tag or sha) ?

5.14.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

none
#### How is this defect reproduced ? 
try to send out data to multiple devices as fast as possible

0xc0170, 2020-03-16:
cc @ARMmbed/mbed-os-pan

adbridge, 2020-03-24:
@CodingGhost you have received the above comment because you have deleted parts of the issue template. Could you please not do this in the future. In this case you deleted '### Description of defect', we will correct it on this occasion.

CodingGhost, 2020-03-24:
noted, thanks.

pan-, 2020-10-05:
@CodingGhost Normally you'd use `onDataSent` to saturate the tx queue -which is shared between all connections - of the controller.  It is a defect that the connection handle isn't reported in the callback.

pan-, 2020-10-21:
@CodingGhost New callbacks have been added in the `GattServer::EventHandler` with  #13734. `EventHandler::onDataSent` accepts a `GattDataSentCallbackParams` which contains the attribute handle and connection handle. Does it solves your problem ?

CodingGhost, 2020-10-21:
I guess this would solve my problem, so thank you for the effort!
But I had to switch to Zephyr because of the power consumption issues with cordio, so I cannot test it for now.

pan-, 2020-10-21:
Thanks for the quick reply. Don't hesitate to give mbed another try once https://github.com/ARMmbed/mbed-os/issues/10669 is resolved. 

I close.

closed 2020-10-21 by pan-