ARMmbed/mbed-os #12605: STM32 EMAC: Stack overflow in DISCO_F769NI Ethernet driver

artokin, 2020-03-10:
### Description of defect
Stack overflow happens in DISCO_F769NI when used as a Border Router.

```
2020-03-10T07:52:11.056Z | D1 <<< DutThread: [DBG ][STE1]: low_level_input len 86
2020-03-10T07:52:11.056Z | D1 <<< DutThread: [DBG ][STE1]: low_level_input no frame
2020-03-10T07:52:11.056Z | D1 <<< DutThread: [INFO][icmp]: Build NA
2020-03-10T07:52:11.056Z | D1 <<< DutThread: [DBG ][STE1]: low_le
2020-03-10T07:52:11.056Z | D1 <-- DutThread: ++ MbedOS Error Info ++
2020-03-10T07:52:11.056Z | D1 <-- DutThread: Error Status: 0x80020125 Code: 293 Module: 2
2020-03-10T07:52:11.057Z | D1 <-- DutThread: Error Message: CMSIS-RTOS error: Stack overflow
2020-03-10T07:52:11.156Z | D1 <-- DutThread: Location: 0x80285EF
2020-03-10T07:52:11.157Z | D1 <-- DutThread: Error Value: 0x1
2020-03-10T07:52:11.157Z | D1 <-- DutThread: Current Thread: stm32_emac_thread Id: 0x20018134 Entry: 0x80214F5 StackSize: 0x400 StackMem: 0x2005FBE0 SP: 0x2007FF34
2020-03-10T07:52:11.157Z | D1 <-- DutThread: For more info, visit: https://mbed.com/s/error?error=0x80020125&tgt=DISCO_F769NI
2020-03-10T07:52:11.158Z | D1 <-- DutThread: -- MbedOS Error Info --
```
It looks like fix https://github.com/ARMmbed/mbed-os/pull/12567 is included in this build as stack size is 0x400.

#### Target(s) affected by this defect ?
DISCO_F769NI

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

fe85cbd903b519b7dcd4b5c104a168ce5a0f7024

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
* Clone git@github.com:ARMmbed/nanostack-border-router.git
* Run `mbed deploy`
* Enable traces: Update `debug-trace` to `true` in `Wisun_Stm_s2lp_RF.json`
* Change Mbed OS to master
* Build nanostack-border-router application for DISCO_F769NI using S2868A1 radio using command: `mbed compile -t GCC_ARM -m DISCO_F769NI -j 0 --app-config ./configs/Wisun_Stm_s2lp_RF.json --profile debug`
* Start the application
* Observe the trace output, the log indicates HardFault

jeromecoutant, 2020-03-10:
Add in your mbed_app.json,
stm32-emac.thread-stacksize = xxxx ?

jeromecoutant, 2020-03-24:
Any update ?
Is this issue still active ?

artokin, 2020-03-24:
The issue is still valid, we have updated our application to use bigger stack:

```
        "DISCO_F769NI": {
            "stm32-emac.thread-stacksize": 1024
        }
```

jeromecoutant, 2020-03-24:
So issue seems solved...?
(no need to patch any files, only application configuration)

artokin, 2020-03-24:
It is now solved in one of our test application by using temporal workaround. 

Other applications will definitely suffer from this same problem and therefore issue should be fixed properly in the driver code.

PR #12728 (2020-04-05): STM32 EMAC : increase thread-stacksize
### Summary of changes 

This fix #12605 

Default thread size is defined to match all configurations.

@artokin 

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2020-04-05 by 0xc0170