ARMmbed/mbed-os #12616: Custom board based on K64F - MK64FN1M0VLQ12 doesn't work

belkirdi, 2020-03-11:
I am not able to setup my custom board based on K64F. I tried all the way without success. This is my error :

Compile [100.0%]: USBPhy_Kinetis.cpp
Link: mbed-os-new_io_motherboard
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
[Error] @0,0: L6218E: Undefined symbol kinetis_init_eth_hardware (referred from BUILD/NEWIOMB/ARMC6/mbed-os/features/netsocket/emac-drivers/TARGET_Freescale_EMAC/kinetis_emac.o).
Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol kinetis_init_eth_hardware (referred from BUILD/NEWIOMB/ARMC6/mbed-os/features/netsocket/emac-drivers/TARGET_Freescale_EMAC/kinetis_emac.o).
Finished: 0 information, 1 warning and 1 error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol kinetis_init_eth_hardware (referred from BUILD/NEWIOMB/ARMC6/mbed-os/features/netsocket/emac-drivers/TARGET_Freescale_EMAC/kinetis_emac.o).
Finished: 0 information, 1 warning and 1 error messages. 

Another issue is I am not able to see my NEWIOMB as USB target on MBED Studio to be capable to run or debug my firmware like how evaluation board does...

I did mbedls to mock my NEWIOMB to 0240 by doing : mbedls --mock 0240:NEWIOMB

When I run the command mbed detect this is what I am getting and as you can is not showing as supported target : 

[mbed] Detected NEWIOMB, port COM10, mounted D:, interface version 0253:
[mbed] Supported toolchains for NEWIOMB

Supported targets: 0
#### Target(s) affected by this defect ?

K64F

#### Toolchain(s) (name and version) displaying this defect ?

ARMC6

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Tool Name | Version | Status |  
ARM Compiler | 6.13 | up to date |  
Clangd | 8.0.1 | up to date |  
Example projects | 1.4.0 | up to date |  
GDB client | 6-2017-q2-update | up to date |  
Git | 2.19.2 | up to date |  
Mbed Library Cache | 1.4.0 | up to date |  
Python tools | 0.3.0 | up to date |  
Debug Packs | 1.0.5 | up to date

#### How is this defect reproduced ? 

This is my config on custom_targets.json

{
    "NEWIOMB": {
        "supported_form_factors": [
            "ARDUINO"
        ],
        "components_add": [
            "SD",
            "FLASHIAP"
        ],
        "core": "Cortex-M4F",
        "supported_toolchains": [
            "ARM",
            "GCC_ARM",
            "IAR"
        ],
        "extra_labels": [
            "Freescale",
            "MCUXpresso_MCUS",
            "KSDK2_MCUS",
            "KPSDK_MCUS",
            "KPSDK_CODE",
            "MCU_K64F",
            "Freescale_EMAC",
            "PSA"
        ],
        "is_disk_virtual": true,
        "macros": [
            "CPU_MK64FN1M0VMD12",
            "FSL_RTOS_MBED",
            "MBED_SPLIT_HEAP",
            "MBED_TICKLESS",
            "TARGET_K64F"
        ],
        "inherits": [
            "Target"
        ],
        "detect_code": [
            "0240"
        ],
        "device_has": [
            "USTICKER",
            "LPTICKER",
            "RTC",
            "CRC",
            "ANALOGIN",
            "ANALOGOUT",
            "EMAC",
            "I2C",
            "I2CSLAVE",
            "INTERRUPTIN",
            "PORTIN",
            "PORTINOUT",
            "PORTOUT",
            "PWMOUT",
            "RESET_REASON",
            "SERIAL",
            "SERIAL_FC",
            "SERIAL_ASYNCH",
            "SLEEP",
            "SPI",
            "SPI_ASYNCH",
            "SPISLAVE",
            "STDIO_MESSAGES",
            "TRNG",
            "FLASH",
            "USBDEVICE",
            "WATCHDOG"
        ],
        "release_versions": [
            "2",
            "5"
        ],
        "device_name": "MK64FN1M0xxx12",
        "bootloader_supported": true,
        "overrides": {
            "network-default-interface-type": "ETHERNET"
        }
    }
}

I copied the files from TARGET_FRDM to a folder TARGET_NEWIOMB in a root folder of the project

belkirdi, 2020-03-11:
if I use this configuration on custom_targets.json : 

{
    "NEWIOMB": {
        "supported_form_factors": [
            "ARDUINO"
        ],
        "components_add": [
            "SD",
            "FLASHIAP"
        ],
        "core": "Cortex-M4F",
        "supported_toolchains": [
            "ARM",
            "GCC_ARM",
            "IAR"
        ],
        "extra_labels": [
            "Freescale",
            "MCUXpresso_MCUS",
            "KSDK2_MCUS",
            "MCU_K64F",
            "FRDM", 
            "KPSDK_MCUS",
            "KPSDK_CODE",
            "Freescale_EMAC",
            "PSA"
        ],
        "is_disk_virtual": true,
        "macros": [
            "CPU_MK64FN1M0VMD12",
            "FSL_RTOS_MBED",
            "MBED_SPLIT_HEAP",
            "MBED_TICKLESS",
            "TARGET_K64F"
        ],
        "inherits": [
            "Target"
        ],
        "detect_code": [
            "0240"
        ],
        "device_has": [
            "USTICKER",
            "LPTICKER",
            "RTC",
            "CRC",
            "ANALOGIN",
            "ANALOGOUT",
            "EMAC",
            "I2C",
            "I2CSLAVE",
            "INTERRUPTIN",
            "PORTIN",
            "PORTINOUT",
            "PORTOUT",
            "PWMOUT",
            "RESET_REASON",
            "SERIAL",
            "SERIAL_FC",
            "SERIAL_ASYNCH",
            "SLEEP",
            "SPI",
            "SPI_ASYNCH",
            "SPISLAVE",
            "STDIO_MESSAGES",
            "TRNG",
            "FLASH",
            "USBDEVICE",
            "WATCHDOG"
[... 27 lines cut ...]
Error: L6200E: Symbol PinMap_RTC multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/PeripheralPins.o).
Error: L6200E: Symbol PinMap_UART_CTS multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/PeripheralPins.o).
Error: L6200E: Symbol PinMap_SPI_SCLK multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/PeripheralPins.o).
Error: L6200E: Symbol PinMap_SPI_SSEL multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/PeripheralPins.o).
Error: L6200E: Symbol PinMap_UART_RTS multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/PeripheralPins.o).
Error: L6200E: Symbol PinMap_UART_TX multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/PeripheralPins.o).
Error: L6200E: Symbol gpio_pinmap multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/PeripheralPins.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/PeripheralPins.o).
Error: L6200E: Symbol crcFast multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/crc.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/crc.o).
Error: L6200E: Symbol g_defaultClockConfigRun multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/fsl_clock_config.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/fsl_clock_config.o).
Error: L6200E: Symbol PHY_GetLinkStatus multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/fsl_phy.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/fsl_phy.o).
Error: L6200E: Symbol crcInit multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/crc.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/crc.o).
Error: L6200E: Symbol crcSlow multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/crc.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/crc.o).
Error: L6200E: Symbol crcTable multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/crc.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/crc.o).
Error: L6200E: Symbol BOARD_BootClockVLPR multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/fsl_clock_config.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/fsl_clock_config.o).
Error: L6200E: Symbol NMI_Handler multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/mbed_overrides.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/mbed_overrides.o).
Error: L6200E: Symbol PHY_GetLinkSpeedDuplex multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/fsl_phy.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/fsl_phy.o).
Error: L6200E: Symbol g_defaultClockConfigVlpr multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/fsl_clock_config.o and BUILD/NE [... cut ...]
Error: L6200E: Symbol PHY_Write multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/fsl_phy.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/fsl_phy.o).
Error: L6200E: Symbol mbed_mac_address multiply defined (by BUILD/NEWIOMB/ARMC6/mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/mbed_overrides.o and BUILD/NEWIOMB/ARMC6/TARGET_NEWIOMB/mbed_overrides.o).
Not enough information to list the image map.
Finished: 1 information, 1 warning and 34 error messages.

belkirdi, 2020-03-11:
> @belkirdi thank you for raising this issue.Please take a look at the following comments:
>
> Could you add some more detail to the description? A good description should be at least 25 words.
>
> NOTE: If [...]

I made the changes and I add another type of issue related to a configuration of the custom_targets.json
Thanks,
Belkirdi

belkirdi, 2020-03-11:
If I use this configuration on custom_targets.json

{
    "NEWIOMB": {
        "inherits": ["K64F"],
        "components_add": ["FLASHIAP"],
        "extra_labels_add": ["K64F"],
        "extra_labels_remove": ["FRDM"],
        "components_remove": ["SD"],
        "supported_form_factors": [],
        "detect_code": ["0240"]
    }
}

I am capable to build the bin successfully but when I copied it manually to a DAPLink I got this fail error :

error: The interface firmware FAILED to reset/halt the target MCU
type: target

error: The interface firmware FAILED to reset/halt the target MCU
type: target

belkirdi, 2020-03-19:
Hello @ciarmcom any update about this issue ?

0xc0170, 2020-03-25:
@ciarmcom is a bot, I'll have a look shortly at this issue.

0xc0170, 2020-03-25:
First, remove `FRDM` from your target (its not freedom board but a custom target). that should fix duplicated symbols.

Second, `kinetis_init_eth_hardware` was not defined - you enabled ethernet but did not provide ethernet driver or reused from FRDM K64F ?

This is not a technical issue but rather a question for https://forum.mbed.com/. Please post there the follow up. Thanks. I'll close this issue.

0xc0170, 2020-03-25:
I've just read the latest messages, so was good to build the application. 

> error: The interface firmware FAILED to reset/halt the target MCU
type: target

Regarding the debugging, ask on the forum to help you. It will help if you provide details about debugger (is it the clone of what is on FRDM K64F -  same MCU for debugging and same connections) ?

closed 2020-03-25 by 0xc0170