ARMmbed/mbed-os #12618: Cellular: EC2x reset module fail

ccchang12, 2020-03-12:
### Description of defect

When the EC2x driver reset module, the soft_power_on() only waits "RDY" message for 5 seconds. But in my test, the EC2x module reset needs 11 second and cause reset fail. Quectel engineer also replies the reset procedure takes about 11 seconds. I think you should increase the timeout.

#### Target(s) affected by this defect ?
K64F + EC21

#### Toolchain(s) (name and version) displaying this defect ?
ARM Compiler 6.12

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.1
Merge: 64853b354f 7b398c0abd

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
For mbed-os-example-cellular, if the cellular module is not in device ready state (e.g. , does not call iface->disconnect() in last test), it will call the soft_power_on().

ccchang12, 2020-03-12:
@ciarmcom , Mbed-OS version is 5.15.1

kivaisan, 2020-03-19:
@ccchang12 Please check the fix proposal.

ccchang12, 2020-03-20:
@kivaisan I think it is OK to fix this issue.

kivaisan, 2020-03-26:
@0xc0170 This issue can be closed as fixed.

kivaisan, 2020-03-26:
@ARMmbed/mbed-os-maintainers This issue can be closed as fixed.

kivaisan, 2020-04-08:
@0xc0170 This issue can be closed.

closed 2020-04-08 by 0xc0170