ARMmbed/mbed-os #12620: Using mbed_trace_prefix_function_set causes crash

kivaisan, 2020-03-12:
### Description of defect

If application sets mbed-trace prefix method with `mbed_trace_prefix_function_set`, system crashes on the first trace print.

This bug was introduced by 0f2b0d1be5783f22ec92cd2487a62809981ad302
cc: @evedon 

#### Target(s) affected by this defect ?
All (?)

#### Toolchain(s) (name and version) displaying this defect ?
Easiest with ARMC6 but seen also with GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
```
static char* prefix_func(size_t ss)
{
    return "";
}
```

And then set prefix method:
```
mbed_trace_init();
mbed_trace_prefix_function_set(&prefix_func);
tr_info("testi");
```

evedon, 2020-03-16:
@kivaisan Could you verify that the fix in https://github.com/ARMmbed/mbed-os/pull/12632 works for you?

kivaisan, 2020-03-16:
@evedon Yes, it fixes the issue.

evedon, 2020-03-16:
> @evedon Yes, it fixes the issue.

Thanks!

closed 2020-03-25 by 0xc0170