ARMmbed/mbed-os #12625: BLE Stack crashes while receiving multiple packets.

prakharbhatt007, 2020-03-13:
### Description of defect
We are trying to send multiple packets using BLE. We have increased the MTU to 240 and we are trying to send packets of 80 bytes. It gives hard-fault after receiving two-three packets. In the callback, there is just a single print statement that we are in a callback. The data we are sending is a byte array.

#### Target(s) affected by this defect ?
DISCO-L475VG-IOT01A

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
same behavior in 5.13.4 and 5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Default Cordio stack that comes with Mbedos

#### How is this defect reproduced? 
In a DISCO Board, a service with write type characteristics, increase the MTU by 240 bytes and try to send multiple packets of data of each length 80 bytes. The mobile app you can use is nrfConnect for Android and the data type should be byte array.

0xc0170, 2020-03-16:
Can you have a look at https://os.mbed.com/docs/mbed-os/v5.15/tutorials/analyzing-mbed-os-crash-dump.html#crash-dump-generation-on-fault-exception and find out what is going on - why it hard faults? It might be app specific.

prakharbhatt007, 2020-03-20:
hey guys
This PR solved the problem that I faced- https://github.com/ARMmbed/mbed-os/pull/12637

0xc0170, 2020-03-25:
Closing as resolved

closed 2020-03-25 by 0xc0170