ARMmbed/mbed-os #12627: Error in rise() method of InterruptIn class for baremetal setting

vsupacha, 2020-03-14:
### Description of defect
The .rise() method of the InterruptIn class does not work as expected. The method behaves like both-edge activation.

#### Target(s) affected by this defect ?
NUCLEO-L053R8 and NUCLEO-F401RE boards.

#### Toolchain(s) (name and version) displaying this defect ?
Arm Compiler 6 included with Mbed Studio 0.9

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 5.15.1
mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 0.9

#### How is this defect reproduced ? 
Button is externally connected with pins PA_7.

InterruptIn SW1(PA_7);
void sw1Handler() {
    cond1 = !cond1;
}
SW1.rise(&sw1Handler); 

Expectation is to toggle the cond1 variable when SW1 is released. But the sw1Handler() is executed for both pressed and released.

JojoS62, 2020-03-14:
@vsupacha are you sure that this problem is not just a bouncing button contact? With a simple button, this cannot be tested. You could connect another DigitalOut with slow toggeling to the InterruptIn to test this more reliable.

vsupacha, 2020-03-14:
@JojoS62 I am using a training board with capacitors to de-bounce buttons. Addition tests found that .rise() method do not work with F401RE with full mbed-os as well. The test procedure is to compare between .rise() and .fall() methods. The .fall() method works perfectly for both PA_9 and PA_10 pins, while .rise() method behaves like rise() + fall() for both pins.

```
InterruptIn SW3(PA_9), SW4(PA_10);
void sw3Handler() {  
    cond3 = !cond3;
}
void sw4Handler() { 
    cond4 = !cond4;
}
SW3.rise(&sw3Handler); 
SW4.fall(&sw4Handler);
```

[image]

JojoS62, 2020-03-14:
the debouncing with a capacitor is not sufficent for an interrupt line. I have tested this on a similar F411 board and it works as expected. PB_12 and PA_9 are connected by a wire.
```cpp
// Blinking rate in milliseconds
#define BLINKING_RATE_MS     (2000)

InterruptIn sw1(PA_9, PullUp);
DigitalOut led1(PC_13);
DigitalOut trigger(PB_12);

void swRiseHandler()
{
    led1 = 0;
    wait_us(50*1000);
    led1 = 1;
}

void swFallHandler()
{
    led1 = 0;
    wait_us(50*1000);
    led1 = 1;
    wait_us(50*1000);
    led1 = 0;
    wait_us(50*1000);
    led1 = 1;
}

int main() 
{
    printf("hello world\n");

    sw1.rise(&swRiseHandler);
    sw1.fall(&swFallHandler);

    while(1) {
        trigger = !trigger;
        ThisThread::sleep_for(BLINKING_RATE_MS);
    }
}
```

vsupacha, 2020-03-15:
@JojoS62 I have added **PullUp** to my code, but rise() method is still triggered by both PRESS and RELEASE. May you please try your code by just only rise() method?
```
InterruptIn SW3(PA_9), SW4(PA_10, PullUp);
SW4.rise(&sw4Handler);
```

JojoS62, 2020-03-15:
it works as expected, with only rise and also with only fall handler.
Have you tried also to test with an output pin to trigger instead of the button?

JojoS62, 2020-03-15:
I have tested it further, connected th button with RC to the interrupt pin. Everything works fine, only one count per press/release. Even the slow slope is working because the interrupt pin has a Schmitt Trigger input.
I have used the current master branch, full os and bare-metal have same behaviour, both are working.

vsupacha, 2020-03-15:
@JojoS62 I have also checked by printing the value of EXTI_RTSR and EXTI_FTSR registers to verify their setting. The values are both ok. So there should be something wrong with the hardware side, even it looks ok from schematic. 

So I would like to close the issue. Thank you very much for your kind support.

closed 2020-03-15 by vsupacha