ARMmbed/mbed-os #12635: Some Mbed OS binaries reference Mbed 6 removed symbols

Patater, 2020-03-16:
### Description of defect

PSA build system symbols are changing for Mbed 6. Binaries shipped with these symbols should be rebuilt and verified.

The following symbols are being removed:
- `COMPONENT_NSPE`
- `COMPONENT_PSA_SRV`
- `COMPONENT_SPE`

```
> git grep COMPONENT_NSPE
Binary file features/netsocket/emac-drivers/TARGET_RDA_EMAC/lwip-wifi/arch/TARGET_RDA/TARGET_UNO_91H/lib/TOOLCHAIN_IAR/libwifi_sta_ap.a matches
Binary file targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_IAR/libublox-odin-w2-driver.a matches
Binary file targets/TARGET_WICED/TOOLCHAIN_IAR/TARGET_MTB_ADV_WISE_1530/libwiced_drivers.a matches
Binary file targets/TARGET_WICED/TOOLCHAIN_IAR/TARGET_MTB_MXCHIP_EMW3166/libwiced_drivers.a matches
Binary file targets/TARGET_WICED/TOOLCHAIN_IAR/TARGET_MTB_USI_WM_BN_BM_22/libwiced_drivers.a matches

> git grep COMPONENT_PSA_SRV
Binary file features/netsocket/emac-drivers/TARGET_RDA_EMAC/lwip-wifi/arch/TARGET_RDA/TARGET_UNO_91H/lib/TOOLCHAIN_ARM_STD/libwifi_sta_ap.ar matches
Binary file features/netsocket/emac-drivers/TARGET_RDA_EMAC/lwip-wifi/arch/TARGET_RDA/TARGET_UNO_91H/lib/TOOLCHAIN_IAR/libwifi_sta_ap.a matches
Binary file targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_IAR/libublox-odin-w2-driver.a matches
Binary file targets/TARGET_WICED/TOOLCHAIN_IAR/TARGET_MTB_ADV_WISE_1530/libwiced_drivers.a matches
Binary file targets/TARGET_WICED/TOOLCHAIN_IAR/TARGET_MTB_MXCHIP_EMW3166/libwiced_drivers.a matches
Binary file targets/TARGET_WICED/TOOLCHAIN_IAR/TARGET_MTB_USI_WM_BN_BM_22/libwiced_drivers.a matches
```

#### Target(s) affected by this defect ?

- Ublox Odin W2
- Wiced driver using targets

#### Toolchain(s) (name and version) displaying this defect ?

- All

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99
Currently removed in a feature branch, found at https://github.com/ARMmbed/mbed-os/tree/feature_twincpu

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli-99.99.99

#### How is this defect reproduced ? 

Grep for deprecated text.

0xc0170, 2020-03-17:
@ARMmbed/team-ublox 
@ARMmbed/mbed-os-ipcore 
@ARMmbed/mbed-os-core 

There is at least one more PR that is requesting rebuilding the binaries. We should coordinate this effort. I mentioned teams who could help to coordinate these or fixed the binaries in our tree.

evedon, 2020-03-17:
@MarceloSalazar is coordinating TARGET_STM32F439xI

MarceloSalazar, 2020-04-08:
RDA and ODIN targets won't be in Mbed 6: https://github.com/ARMmbed/mbed-os/pull/12775