ARMmbed/mbed-os #12647: Cypress: FPGA CI: SPI: regression from commit b2dad08 to 4b6c844

yarbcy, 2020-03-18:
### Description of defect
Cypress: FPGA CI: SPI: regression from commit b2dad08 to 4b6c844

PASS: b2dad0838778d16a6a73067b9ce01c85c1774f3c
FAIL: 4b6c844be9574ae9177eaf28549f505083d62356

Diff:
https://github.com/ARMmbed/mbed-os/commit/4b6c844be9574ae9177eaf28549f505083d62356#diff-c68433fe07da3e9c69a88390f75741db

Fail log:
***Testing all SPI peripherals***
[1584106281.79][CONN][RXD] 1080098816 - Could not find pins to test peripheral
[1584106281.95][CONN][RXD] 1080164352 - peripheral tested on port: peripheral=(1080164352) MOSI=(A0) MISO=(A1) SCLK=(A2) ...:290::FAIL: Expected 0 Was 256
[1584106282.03][CONN][RXD] >>> 'SPI - basic test': 0 passed, 1 failed with reason 'Assertion Failed'
[1584106282.03][CONN][INF] found KV pair in stream: {{__testcase_finish;SPI - basic test;0;1}}, queued...
[1584106282.11][CONN][RXD]
[1584106282.11][CONN][RXD] >>> Test cases: 3 passed, 1 failed with reason 'Assertion Failed'
[1584106282.19][CONN][RXD] >>> TESTS FAILED!

#### Target(s) affected by this defect ?
CY8CKIT_062_WIFI_BT

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
4b6c844be9574ae9177eaf28549f505083d62356

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
Always
[GT-FT-KIT-062-WIFI-BT-FPGA-GCC.log](https://github.com/ARMmbed/mbed-os/files/4349376/GT-FT-KIT-062-WIFI-BT-FPGA-GCC.log)

yarbcy, 2020-03-18:
@mprse @0xc0170 Please take a look.

I see assert fail in line 159, file spi/main.cpp:
[image]

yarbcy, 2020-03-20:
@0xc0170 Can you please take a look?

yarbcy, 2020-03-20:
@adbridge ?

adbridge, 2020-03-20:
@yarbcy I'll see if I can find out who knows about the SPI stuff and if they have time to take a look

mprse, 2020-03-20:
@yarbcy After the test has been extended the `CS` is now handled by default manually:
https://github.com/ARMmbed/mbed-os/blob/03dab7f9a0249d75ae667f143e78a824dbf9770c/TESTS/mbed_hal_fpga_ci_test_shield/spi/main.cpp#L254-L256

I believe only this impacts the basic test case. In the previous version, only hardware CS handling was used in testing.

0xc0170, 2020-03-22:
@mprse So what should a target do ? Is `SS_DEASSERT` not defined properly as the log states above? `Could not find pins to test peripheral` ?

mprse, 2020-03-23:
The pins have been found, but the test has failed:
```
1080164352 - peripheral tested on port: peripheral=(1080164352) MOSI=(A0) MISO=(A1) SCLK=(A2) ...:290::FAIL: Expected 0 Was 256
```

I suggest checking if the `HW SSEL` test case passes (comment out other test cases than `SPI - hardware ss handling`). If it is passing, then check which pin has been selected for `SSEL` in the basic test and check what can be wrong with it.

yarbcy, 2020-03-24:
Closed because https://github.com/ARMmbed/mbed-os/issues/12687 is more complex.

closed 2020-03-24 by yarbcy