ARMmbed/mbed-os #12675: COMPONENT_FPGA_CI_TEST_SHIELD: FLASH_SECTOR_SIZE macro redefinition

jeromecoutant, 2020-03-23:
### Description of defect

FLASH_SECTOR_SIZE is already used by several targets:

https://github.com/ARMmbed/mbed-os/blob/539cf3fa6396dd3a6b0377e715d7b0d40c7af4c4/components/testing/COMPONENT_FPGA_CI_TEST_SHIELD/MbedTester.cpp#L33

````
targets/TARGET_ARM_FM/TARGET_FVP_MPS2/flash_api.c:#define FLASH_SECTOR_SIZE 0x1000
targets/TARGET_ARM_SSG/TARGET_CM3DS_MPS2/flash_api.c:#define FLASH_SECTOR_SIZE 0x1000
targets/TARGET_RENESAS/TARGET_RZ_A1XX/TARGET_GR_LYCHEE/mbed_drv_cfg.h:#define FLASH_SECTOR_SIZE          4096           /**< Flash Memory sector size (interleaving off) */
targets/TARGET_RENESAS/TARGET_RZ_A1XX/TARGET_RZ_A1H/mbed_drv_cfg.h:#define FLASH_SECTOR_SIZE          4096           /**< Flash Memory sector size (interleaving off) */
targets/TARGET_Realtek/TARGET_AMEBA/TARGET_MCU_RTL8195A/ota_api.h:#define FLASH_SECTOR_SIZE    0x1000
targets/TARGET_Realtek/TARGET_AMEBA/sdk/soc/realtek/8195a/fwlib/rtl8195a/rtl8195a.h:#define FLASH_SECTOR_SIZE                           0x1000
targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H743xI/device/stm32h743xx.h:#define FLASH_SECTOR_SIZE  0x00020000UL        /* 128 KB */
targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H747xI/device/stm32h747xx.h:#define FLASH_SECTOR_SIZE  0x00020000UL        /* 128 KB */
````

I think you should rename it

Thx

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ? 

Compilation warning:
````
Compile [  1.4%]: MbedTester.cpp
[Warning] MbedTester.cpp@33,9: 'FLASH_SECTOR_SIZE' macro redefined [-Wmacro-redefined]
````

0xc0170, 2020-03-24:
This is one of the reasons why I would use regular variable in unnamed namespace rather  than macro. We should fix this

cc @mprse @maciejbocianski @fkjagodzinski

jeromecoutant, 2020-06-08:
ping

MarceloSalazar, 2021-03-12:
Fixed in #14076

closed 2021-03-12 by MarceloSalazar