ARMmbed/mbed-os #12678: QSPI driver double call qspi_init()

dustin-crossman, 2020-03-23:
### Description of defect
In QSPI::_initialize mbed::QSPI will call the target specific qspi_init() once. Later, if QSPI::_acquire() gets called (which it does in QSPI::read(), QSPI::write(), etc) qspi_init() will be called again. A better description of the issue is here: https://github.com/ARMmbed/mbed-os/issues/11230 (Also note that there is still no documentation regarding this).

There seems to be 2 ways to resolve this:
1. Completely remove _acquire() and the drivers concept of ownership completely.
2. Maintain the drivers concept of ownership and explicitly document the implications and requirements in the QSPI porting guide.

In the meantime I believe there is a way to partially mitigate the issue:
In QSPI::_intialize (and _initialize_direct) the _initialized field is being set to true but the _owner field is not. Since _owner is false upon the first read/write call, _acquire() and therefore _initialize() will be called again even if no other QSPI object acquired ownership in the meantime. I think this is a bug regardless of the approach taken to fix this issue and is an easy fix.

I'm happy to put in a PR for the mitigation but I'd also like to see this resolved properly one way or the other.

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?

Latest master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
Any use of QSPI driver will reproduce this defect.

0xc0170, 2020-03-23:
cc @ARMmbed/mbed-os-hal 

> In QSPI::_intialize (and _initialize_direct) the _initialized field is being set to true but the _owner field is not. Since _owner is false upon the first read/write call, _acquire() and therefore [...]

Lets fix this.

>  #11230

That was a question but looks like a bug that should be opened and fixed.

dustin-crossman, 2020-03-23:
@0xc0170 Ok. I put in a PR for the partial fix: https://github.com/ARMmbed/mbed-os/pull/12680
Should we reopen that old issue or continue here?

0xc0170, 2020-03-24:
@adbridge See ^^ duplication

adbridge, 2020-03-24:
#2603 closed as duplicate and link deleted

0xc0170, 2020-04-03:
> @0xc0170 Ok. I put in a PR for the partial fix: #12680

It was merged. This can be closed or what's left @dustin-crossman ?

dustin-crossman, 2020-04-03:
@0xc0170 #12680 only partially fixes the double init problem. It would be good to figure out a proper resolution to the issue, especially since this isn't the first time it's been reported.

closed 2021-03-26 by ciarmcom