ARMmbed/mbed-os #12687: SPI: FPGA CI: test regression

yarbcy, 2020-03-24:
### Description of defect

SPI: FPGA CI: test regression

**First regression:**
From commit fd05ebf96ddcee3e231dd602abfd176b3d4ed2c7 (PASS) to ba12228556bbac4b8e0b9819a94a946513babecc (FAIL)
Diff: github.com/ARMmbed/mbed-os/compare/fd05ebf96ddcee3e231dd602abfd176b3d4ed2c7...ba12228556bbac4b8e0b9819a94a946513babecc

**Proposed fix:**
Line 130 -136:
   //if (init_direct) {
        //const spi_pinmap_t pinmap = get_spi_pinmap(mosi, miso, sclk, ssel);
        //spi_init_direct(&spi, &pinmap);
    //} else {
        spi_init(&spi, mosi, miso, sclk, ssel);
    //}
**After this code change - all 12 test cases PASSED.**

**Second regression:**
From commit b2dad08 to master.
Diff: 
https://github.com/ARMmbed/mbed-os/compare/b2dad08...master#files_bucket

Fix (above) doesn't help in this case.

Fail log:
[FPGA_CI_CY8CKIT_062_WIFI_BT.txt](https://github.com/ARMmbed/mbed-os/files/4375373/FPGA_CI_CY8CKIT_062_WIFI_BT.txt)

#### Target(s) affected by this defect ?
CY8CKIT_062_WIFI_BT

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
See above.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
Always

yarbcy, 2020-03-24:
@adbridge Please review.

adbridge, 2020-03-24:
@yarbcy 
'What version of Mbed-os are you using (tag or sha) ?
See above.

This will fail the automated checks. Please follow the guidelines and put a specific version in this field.

Also

'How is this defect reproduced ? '

This should be a list of steps that you run to produce this error .

yarbcy, 2020-03-24:
> @yarbcy
> 'What version of Mbed-os are you using (tag or sha) ?
> See above.
> 
> This will fail the automated checks. Please follow the guidelines and put a specific version in this field.
> 
> Also
> 
> 'How is this defect reproduced ? '
> 
> This should be a list of steps that you run to produce this error .

Done.

0xc0170, 2020-03-25:
cc @mprse

0xc0170, 2020-03-26:
> cc @mprse

We'll need to allocate this to different person.

0xc0170, 2020-04-02:
cc @jamesbeyond Can you help with this one?

jamesbeyond, 2020-04-02:
> cc @jamesbeyond Can you help with this one?
Yep, I am assigning this one to me

jamesbeyond, 2020-04-06:
Hi @yarbcy, 

I think the main issue caused by we introduced the static pinmap function to mbed-os for saving flash usage in run time. [Static Pinmap](https://github.com/ARMmbed/mbed-os/blob/master/docs/design-documents/hal/0003-static-pinmap-extension.md)
the test for the FPGA shield tests trends to use the static pinmap. But the target `CY8CKIT_062_WIFI_BT` and all the cypress target did not have this feature enabled. so that the reason the tests are failing.

So I would suggest to cypress team to enable the static pinmap following by design docs: [Static Pinmap](https://github.com/ARMmbed/mbed-os/blob/master/docs/design-documents/hal/0003-static-pinmap-extension.md). the only work that need to be done is spilt the exist `spi_init()` function into a`spi_init()` and a `spi_init_direct()` function. then the tests would be passed.