ARMmbed/mbed-os #12688: Add CI for CY8CKIT_062_WIFI_BT - FPGA_CI

yarbcy, 2020-03-24:
### Description of defect
Add Continious integration test for CY8CKIT_062_WIFI_BT - FPGA_CI (Test Shield).
Test regresion was detected by https://github.com/ARMmbed/mbed-os/issues/12687
SPI test cases changed from PASS state to FAIL. Ii is because of new code change to file: "mbed-os\TESTS\mbed_hal_fpga_ci_test_shield\spi\main.cpp"

#### Target(s) affected by this defect ?
CY8CKIT_062_WIFI_BT 

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?
https://github.com/ARMmbed/mbed-os/commit/5d64e558801940b3be547c3d059f28eaabc23f95

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
Always

yarbcy, 2020-03-24:
@adbridge Please review.

adbridge, 2020-03-24:
@yarbcy 
'How is this defect reproduced ?'
This should be a list of steps run to reproduce the failure

yarbcy, 2020-03-24:
> @yarbcy
> 'How is this defect reproduced ?'
> This should be a list of steps run to reproduce the failure

Done.

0xc0170, 2020-03-31:
@yarbcy This is the request to add a new target to CI nightly? That is how I understand the initial report.

yarbcy, 2020-04-02:
> @yarbcy This is the request to add a new target to CI nightly? That is how I understand the initial report.

Yes it is.

adbridge, 2020-04-08:
@yarbcy where does your commit hash come from? The script looks for a PR that has been merged to master that contains that hash. If it cannot find one then it raised the above comment...

yarbcy, 2020-04-23:
Fixed.

0xc0170, 2020-04-23:
@ARMmbed/mbed-os-test Please review

jamesbeyond, 2020-07-16:
We are waiting for the new order of FPGA shileds

closed 2021-10-20 by ciarmcom