ARMmbed/mbed-os #12695: DISCO_L475VG_IOT01A:  failing features-storage-tests-kvstore-direct_access_devicekey_test

0xc0170, 2020-03-25:
### Description of defect

DISCO_L475VG_IOT01A `features-storage-tests-kvstore-direct_access_devicekey_test` fail often since the last week (Thursday/Friday). It is blocking now most of our PRs (not 100 percent). See the discussion about this issue here https://github.com/ARMmbed/mbed-os/pull/12632#issuecomment-603193646 (also logs for the failures there).

Nightly does not fail. Latest nightly as of today: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/495/greentea-test/137/FAIL/.  From the logs in the PASS folder: `| DISCO_L475VG_IOT01A-ARMC6 | DISCO_L475VG_IOT01A | features-storage-tests-kvstore-direct_access_devicekey_test                  | OK     | 29.78              | default     |`

I checked older nightly tests (few days back), this failure is not there as in PRs.
This means only PRs are affected, I am not certain what is different between nightly and PRs.

It's only ARM (haven't yet seen GCC ARM failures).

#### Target(s) affected by this defect ?

DISCO_L475VG_IOT01A 

#### Toolchain(s) (name and version) displaying this defect ?

ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not applicable in this case

#### How is this defect reproduced ? 

`mbed test -m DISCO_L475VG_IOT01A -t arm -n features-storage-tests-kvstore-direct_access_devicekey_test`

0xc0170, 2020-03-25:
As it affects quite number of PRs, I'll create a label to track them.

cc @mbed-os-test @ARMmbed/mbed-os-storage @ARMmbed/mbed-os-core

jarlamsa, 2020-03-25:
As @SeppoTakalo mentioned in couple of the PR's that it might be faulty HW. Checking from the logs of 2 different failing PR's, the device used is the same, so I am quite certain that the particular HW is the culprit.

0xc0170, 2020-03-25:
PRs affected can be viewed here: 
https://github.com/ARMmbed/mbed-os/pulls?q=is%3Apr+is%3Aopen+label%3A%22needs%3A+CI+device+key+failure%22

0xc0170, 2020-03-25:
One PR got green, will wait for more PRs to complete

0xc0170, 2020-03-26:
Looks like it was fixed, I'll close this.

0xc0170, 2020-04-02:
One of the latest findings: https://mbed-os.mbedcloudtesting.com/job/mbed-os-ci_greentea-test/3975/testReport/junit/DISCO_L475VG_IOT01A-ARMC6/features-storage-tests-kvstore-direct_access_devicekey_test/DISCO_L475VG_IOT01A_ARM___Testing_direct_access_to_devicekey_with_tdb_over_flashiap_default_placement/ (internal link).

or

https://mbed-os.mbedcloudtesting.com/job/mbed-os-ci_greentea-test/3978/testReport/junit/DISCO_L475VG_IOT01A-ARMC6/features-storage-tests-kvstore-direct_access_devicekey_test/DISCO_L475VG_IOT01A_ARM___Testing_direct_access_to_devicekey_with_tdb_over_flashiap_default_placement/ (internal link)

The log from there:

```
[1585770388.79][HTST][INF] host test detected: default_auto
[1585770388.88][CONN][INF] found KV pair in stream: {{__testcase_name;Testing direct access to devicekey with tdb over flashiap default placement}}, queued...
[1585770388.99][CONN][INF] found KV pair in stream: {{__testcase_name;Testing direct access to devicekey with tdb over last two sectors}}, queued...
[1585770389.09][CONN][RXD] 
[1585770389.09][CONN][INF] found KV pair in stream: {{__testcase_name;Testing direct access to injected devicekey }}, queued...
[1585770389.19][CONN][RXD] >>> Running case #1: 'Testing direct access to devicekey with tdb over flashiap default placement'...
[1585770389.29][CONN][RXD] Test Direct Access To DeviceKey Test Entire FlashIAP Default Address
[1585770389.29][CONN][INF] found KV pair in stream: {{__testcase_start;Testing direct access to devicekey with tdb over flashiap default placement}}, queued...
[1585770389.39][CONN][RXD] :130::FAIL: Expected 0 Was 283
[1585770389.39][CONN][RXD] :133::FAIL: Expected 0 Was 270
[1585770389.39][CONN][RXD] 
[1585770389.49][CONN][RXD] ++ MbedOS Fault Handler ++
[1585770389.49][CONN][RXD] 
[1585770389.49][CONN][RXD] FaultType: HardFault
```

I can still see it seldomly on PRs. Although this time there is

0xc0170, 2020-04-03:
Also over night another http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci/PR-12702/artifacts/12702/2/greentea-test/3982/

@SeppoTakalo Please review

0xc0170, 2020-04-08:
@VeliMattiLahtela The boards were put into maintenance. Anything to do for this issue?

0xc0170, 2020-04-09:
Still failing however, one of the latest PR: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci/PR-12271/artifacts/12271/5/greentea-test/4017/FAIL/

teetak01, 2020-04-09:
Just wondering, the KVstore is located at the end of the board in these tests right? @SeppoTakalo @VeijoPesonen  ?

Thus normal flashing with a test-binary will not erase the kvstore. Do the tests work if there is garbage in the device, or will they randomly fail if the device is not purely green?

Is there some cleanup test as first to make sure the board is in clean state?

Now I am not very familiar with these tests, but similar issues have been considered in client-side testing to ensure pre-existing KVstore data on shared devices cannot cause issues.

evedon, 2020-11-09:
@0xc0170 Can you update the ticket or close?

0xc0170, 2020-11-27:
Closing, I haven't seen the failures in the last days.

closed 2020-11-27 by 0xc0170