ARMmbed/mbed-os #12700: mbed export on windows generates ASM_FLAGS with wrong slashes

shuopeng-deng, 2020-03-25:
### Description of defect
mbed export -i eclipse_gcc_arm -m CY8CPROTO_062S3_4343W
Generates makefile with following lines:
ASM_FLAGS += -include
ASM_FLAGS += .\mbed_config.h

Attemping to compile using the bash tools such as cygwin or gitbash will result in following error:
cc1.exe: fatal error: .mbed_config.h: No such file or directory

The OBJECTS and INCLUDE_PATHS are generated with forward slash.
The ASM_FLAGS should be generated with forward slash as well.

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
GCC

#### What version of Mbed-os are you using (tag or sha) ?
5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.2

#### How is this defect reproduced ? 
1. mbed import mbed-os-example-blinky
2. mbed export -i eclipse_gcc_arm -m CY8CPROTO_062S3_4343W
3. See attached generated makefile line 1754 to 2092
[Makefile.txt](https://github.com/ARMmbed/mbed-os/files/4382856/Makefile.txt)

0xc0170, 2020-03-26:
cc @ARMmbed/mbed-os-tools

0xc0170, 2020-04-27:
@shuopeng-deng Can you change this function located in tools/export/makefile/__init__.py

```
    def format_flags(self):
        """Format toolchain flags for Makefile"""
        flags = {}
        for k, v in self.flags.items():
            if k in ['c_flags', 'cxx_flags']:
                flags[k] = list(map(lambda x: x.replace('"', '\\"'), v))
            else:
                flags[k] = v

        return flags
```

It should include `asm_flags` to format as well. Please send PR fixing this (once confirm this fixes the issue).

shuopeng-deng, 2020-05-01:
That function only escapes double-quotes by replaces " with \\".
It does not replace forward slash with backslashes.