ARMmbed/mbed-os #12709: Ethernet on a custom board failing

Albis-dev, 2020-03-28:
### Description of defect
We made a custom board based on NUCLEO-F429ZI but with a different PHY chip (DP83848CVV). It seems to successfully initialize itself. Here's the code that works without any issue. 

```
EthernetInterface net;

#define IP "192.168.0.3"
#define MASK "255.255.255.0"
#define GATEWAY "192.168.0.2"

int InitNetwork() {
    SocketAddress ip = SocketAddress(IP, 0);
    SocketAddress mask = SocketAddress(MASK, 0);
    SocketAddress gateway = SocketAddress(GATEWAY, 0);
    
    if(net.set_network(ip,mask,gateway) != 0) {
        pc.printf("Failed Establishing Ethernet Interface\n");
        return -1;
    }
    
    if(net.connect() != 0) {
        pc.printf("Failed connecting Ethernet\n");
        return -1;
    }
    return 0;
}
```
We connect the board and a PC through Ethernet and run a small test. Basically, the PC sends some number through UDP. The board receives the data and shows it through serial communication.

```
void receive()
{
    SocketAddress receive;
    UDPSocket socket(&net);
    int bind = socket.bind(8080);
    pc.printf("bind return: %d\n", bind);
 
    uint16_t buffer;
    while (true)
    {
        pc.printf("\nWait for packet...\n");
        socket.recvfrom(&receive, &buffer, sizeof(buffer));
        
        switch(buffer) {
            case 0x0001:
                pc.printf("That's one.");
                break;
            default:
                pc.printf("Unable to Interpret the code");
                break;
        }
    }
}
```
This works well when we're using NUCLEO-F429ZI. However when it comes to our custom board, our board does not receive any data from PC. Weirdly enough, It does executes `InitNetwork()` without a problem. Also we were able to find out its MAC address using `get_mac_address()`. `get_connection_status()` returns `NSAPI_STATUS_GLOBAL_UP`.

If there's anything I can provide, please let me know!
Thanks for your time and effort.

#### Target(s) affected by this defect ?
None

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Compiler

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
None

#### How is this defect reproduced ? 
As stated above.

jeromecoutant, 2020-03-30:
Hi
Could you check master branch with #12464 integrated?
Then check values in 
https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/emac-drivers/TARGET_STM/mbed_lib.json

Albis-dev, 2020-03-30:
Hi @jeromecoutant 
What you are suggesting is that I should merge that specific PR into the current mbed-os master branch and set the values in mbed_lib.json appropriately, right?

0xc0170, 2020-03-30:
> What you are suggesting is that I should merge that specific PR into the current mbed-os master branch and set the values in mbed_lib.json appropriately, right?

As indicated, you tested mbed-os-5.15.1 version. If you can retest latest master that include also 12464 PR?

Albis-dev, 2020-03-30:
Ah, acknowledged. Will do!

Albis-dev, 2020-03-31:
With 92cdcfb30286575c188871c437b10006ac811457, EthernetInterface::connect() now returns -3004 which is `NSAPI_ERROR_NO_CONNECTION`.

jeromecoutant, 2020-03-31:
Did you update values in
https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/emac-drivers/TARGET_STM/mbed_lib.json
with your PHY values?

Albis-dev, 2020-03-31:
> Did you update values in
> https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/emac-drivers/TARGET_STM/mbed_lib.json
> with your PHY values?

I've updated the following according to the datasheet :
1. "eth-phy-status-register"
2. "eth-phy-speed-status"
3. "eth-phy-duplex-status"

Is there anything else should be modified?

Albis-dev, 2020-04-02:
After changing "eth-phy-address" to "1" `InitNetwork()` works. The main issue still persists though.
I examined the network using Wireshark and found that It never announces its IP through ARP broadcast (ARP Announcement). However It seems that "ARP Probe" is continuing normally.

Albis-dev, 2020-04-04:
Customising `void HAL_ETH_MspInit(ETH_HandleTypeDef *heth)` fix this issue. Closing and thanks for assisting!

closed 2020-04-04 by Albis-dev

jeromecoutant, 2020-04-06:
> Customising `void HAL_ETH_MspInit(ETH_HandleTypeDef *heth)` fix this issue. Closing and thanks for assisting!

Yes, you need to implement your HAL_ETH_MspInit for your own target

Other solution is to enable USE_USER_DEFINED_HAL_ETH_MSPINIT,
and implement HAL_ETH_MspInit in your application.