ARMmbed/mbed-os #12710: Higher Baudrate of Component ESP8266 WiFi

cyliangtw, 2020-03-29:
### Description of defect

According to esp8266-driver mbed_lib.json, to adjust baudrate as 460800 instead of default 115200. The ESP8266 driver can't work except rollback 115200.
However, I tried the direct control of esp8266 module by AT command through pass-through, it could get successful return by `"AT+UART_DEF=460800,8,1,0,3"`. Thus, this esp8266 module is fine for 460800 baudrate, the issue seems esp8266 driver manner.

Pass-through AT log as below:
```
AT+UART_DEF=460800,8,1,0,3

OK
AT+GMR
AT version:1.7.0.0(Aug 16 2018 00:57:04)
SDK version:3.0.0(d49923c)

```
The snippet of my mbed_app.json :
```
            "target.network-default-interface-type" : "WIFI",
	    "esp8266.serial-baudrate"               : 460800,
            "esp8266.tx"                            : "PH_8",
            "esp8266.rx"                            : "PH_9",
            "esp8266.rts"                           : "A2",
            "esp8266.cts"                           : "A3",
            "esp8266.rst"                           : "PH_3",
            "esp8266.provide-default"               : true

```

0xc0170, 2020-03-30:
cc @ARMmbed/mbed-os-ipcore

michalpasztamobica, 2020-03-30:
I checked the code and docs briefly, but I see nothing that would stop you from using higher baud rate on ESP8266.
I also compiled some code for K64F with baud set to 460800 and the communication went on fine (at least the board got configured and connected to Wi-Fi normally).
What exactly is the issue that you're seeing? What is the platform you're using? Does it support high-speed UART?

cyliangtw, 2020-03-30:
In ESP8266 constructor, to set the assigned baud by
` _serial.set_baud(MBED_CONF_ESP8266_SERIAL_BAUDRATE);`. 
However, the default baud of ESP8266 module is 115200, so MCU can't send AT command correctly to the ESP8266 module after constructor set serial baud as 460800.
Is your ESP8266 module default baud 460800 ?

michalpasztamobica, 2020-03-30:
This macro `MBED_CONF_ESP8266_SERIAL_BAUDRATE` is being set based on the setting in in mbed_app.json. Please check `BUILD/<platform>/<compiler>/mbed_config.h` file. With the `mbed_app.json` entry which you provided you should have something like this:
```
#define MBED_CONF_ESP8266_SERIAL_BAUDRATE          460800     // set by application[K64F]
```

cyliangtw, 2020-03-31:
@michalpasztamobica ,  yes, both of my mbed_app.json and mbed_config.h are baudrate 460800. 
But ESP-8266 firmware default setting is 115200 & no flow control, the serial obj with 460800 baud can't communicate with ESP8266 module. 
My solution is to set 115200 as ESP8266_DEFAULT_BAUDRATE and to switch the specific baudrate  in at_available().

cyliangtw, 2020-04-01:
@michalpasztamobica,  I enabled debug and traced AT command as while suffered this issue in OS 5.15 . Today, I also confirmed the same symptom in OS 6.0 beta. So, after I verify the solution OK, I will send one PR.

closed 2020-04-01 by cyliangtw

michalpasztamobica, 2020-04-01:
@cyliangtw , your contribution will be most welcome, thanks! Looking forward to the PR.