ARMmbed/mbed-os #12718: Printf/sprintf not printing leading zeros

pilotak, 2020-03-30:
### Description of defect

Using a following code compiled against os5.15 will produce following output (which is expected)
```
0C
90
03
00
16
57
4A
48
38
04
37
20
result[24]: 0C90030016574A4838043720
```

but same code with os6.0.0-alpha3 produces
```
C
90
3
0
16
57
4A
48
38
4
37
20
result[24]: C
```

#### Target(s) affected by this defect ?
Maybe all, i tried it on STM32F303RE

#### Toolchain(s) (name and version) displaying this defect ?
 gcc-arm-none-eabi-9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.0.0-alpha-3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
cli: 1.10.1

#### How is this defect reproduced ? 

```cpp
#include "mbed.h"

const uint8_t id[12] = {
    0x0c, 0x90, 0x03, 0x00, 0x16, 0x57, 0x4a, 0x48, 0x38, 0x04, 0x37, 0x20
};

int main() {
    char str[50] = {0};
    int str_len = 0;

    for (int i = 0; i < 12; i++) {
        printf("%02X\n", id[i]);
        str_len += snprintf(str + str_len, sizeof(str), "%02X", id[i]);
    }

    printf("result[%i]: %s\n", str_len, str);

    return 0;
}
```

0xc0170, 2020-03-31:
The difference there is enabling minimal printf (between tested versions), isn't it?

cc @ARMmbed/mbed-os-core

evedon, 2020-03-31:
The minimal printf library is enabled by default in Mbed os 6.0.0-alpha3. Not all format speficiers are supported, in particular width formatter. If your application requires them then you can use the standard printf library by setting "target.printf_lib" to "std", see https://github.com/ARMmbed/mbed-os/tree/master/platform/source/minimal-printf

pilotak, 2020-03-31:
OK, it would be a good idea to put it into documentation somewhere.
Is there any plan to support it in the future?

evedon, 2020-03-31:
> Is there any plan to support it in the future?

There is no plan to support all format specifiers as the main aim of the minimal-printf library is to have a small memory footprint. We recognise that the width specifier is widely used but we believe that this is mostly for pretty printing.

pilotak, 2020-04-01:
That's a shame, it would be a nice feature that could be turned on/off if needed like support for 64bit. As you say many people use it.

closed 2020-04-02 by pilotak

0xc0170, 2020-04-02:
> OK, it would be a good idea to put it into documentation somewhere.

@evedon are there docs where we can find this limitation? I only found this in our docs: 

> Mbed OS provides the minimal-printf library, which offers a subset of the printf features (not all format specifiers are supported). You can also achieve further flash savings if your application does not require 64-bit integers or floating point printing.