ARMmbed/mbed-os #12720: Errors while compiling test cases for ODIN W2

qasim-ublox, 2020-03-30:
### Description of defect
Build failure is occurring while compiling tests with toolchains GCC, ARM, IAR.
Errors can be seen below. Detailed build logs are also attached
[arm_build_logs.txt](https://github.com/ARMmbed/mbed-os/files/4404294/arm_build_logs.txt)
[gcc_build_logs.txt](https://github.com/ARMmbed/mbed-os/files/4404295/gcc_build_logs.txt)
[iar_build_logs.txt](https://github.com/ARMmbed/mbed-os/files/4404296/iar_build_logs.txt)

`GCC:`
Build failures:
  * UBLOX_EVK_ODIN_W2::GCC_ARM::MBED-OS-FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM_RECOVERY-RESILIENCE_FUNCTIONAL
        Building project resilience_functional (UBLOX_EVK_ODIN_W2, GCC_ARM)
        Scan: GCC_ARM
        Scan: resilience_functional
        Scan: COMMON
        
  * UBLOX_EVK_ODIN_W2::GCC_ARM::MBED-OS-TESTS-NETWORK-MULTIHOMING
        Building project multihoming (UBLOX_EVK_ODIN_W2, GCC_ARM)
        Scan: GCC_ARM
        Scan: multihoming
        Compile [ 25.0%]: main.cpp
        [Error] unity_internals.h@669,139: invalid cast from type 'SocketAddress' to type '_U_SINT' {aka 'long long int'}
        [Error] unity_internals.h@669,158: invalid cast from type 'SocketAddress' to type '_U_SINT' {aka 'long long int'}
        [Error] unity_internals.h@669,139: invalid cast from type 'SocketAddress' to type '_U_SINT' {aka 'long long int'}
        [Error] unity_internals.h@669,158: invalid cast from type 'SocketAddress' to type '_U_SINT' {aka 'long long int'}

`ARM:`
Build failures:
  * UBLOX_EVK_ODIN_W2::ARMC6::MBED-OS-FEATURES-STORAGE-FILESYSTEM-LITTLEFS-TESTS-FILESYSTEM_RECOVERY-RESILIENCE_FUNCTIONAL
        Building project resilience_functional (UBLOX_EVK_ODIN_W2, ARMC6)
        Scan: ARMC6
        Scan: resilience_functional
        Scan: COMMON

  * UBLOX_EVK_ODIN_W2::ARMC6::MBED-OS-TESTS-NETWORK-MULTIHOMING
        Building project multihoming (UBLOX_EVK_ODIN_W2, ARMC6)
        Scan: ARMC6
        Scan: multihoming
        Compile [ 25.0%]: main.cpp
        [Warning] LWIPStack.h@300,27: 'get_ip_address' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
        [Error] main.cpp@84,5: cannot convert 'SocketAddress' to '_U_SINT' (aka 'long long') without a conversion operator
        [Error] main.cpp@84,5: cannot convert 'SocketAddress' to '_U_SINT' (aka 'long long') without a conversion operator
        [Error] main.cpp@115,9: cannot convert 'SocketAddress' to '_U_SINT' (aka 'long long') without a conversion operator
        [Error] main.cpp@115,9: cannot convert 'SocketAddress' to '_U_SINT' (aka 'long long') without a conversion operator

`IAR:`
Building library mbed-build (UBLOX_EVK_ODIN_W2, IAR)
Scan: test
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
Compile [ 69.9%]: SocketAddress.cpp
[Error] SocketAddress.cpp@47,18: [Pe304]: no instance of overloaded function "std::unique_ptr<_Ty [], _Dx>::reset [with _Ty=char, _Dx=std::default_delete<char []>]" matches the argument list
[Error] SocketAddress.cpp@76,0: [Pe304]: no instance of overloaded function "std::unique_ptr<_Ty [], _Dx>::reset [with _Ty=char, _Dx=std::default_delete<char []>]" matches the argument list
[ERROR]
      _ip_address.reset();
                  ^
"D:\mbed5\TE_OM2-405\m\test\mbed-os\features\netsocket\SocketAddress.cpp",47  Error[Pe304]: no instance of overloaded function "std::unique_ptr<_Ty [], _Dx>::reset [with _Ty=char, _Dx=std::default_delete<char []>]" matches the argument list
            object type is: std::unique_ptr<char []>

      _ip_address.reset();
                  ^
"D:\mbed5\TE_OM2-405\m\test\mbed-os\features\netsocket\SocketAddress.cpp",76  Error[Pe304]: no instance of overloaded function "std::unique_ptr<_Ty [], _Dx>::reset [with _Ty=char, _Dx=std::default_delete<char []>]" matches the argument list
            object type is: std::unique_ptr<char []>

Failed to build library

#### Target(s) affected by this defect ?
Ublox Odin W2

#### Toolchain(s) (name and version) displaying this defect ?
Arm Compiler 6.13
GNU Arm Embedded version 9 (9-2019-q4-major)
IAR Embedded Workbench 8.32.1

#### What version of Mbed-os are you using (tag or sha) ?

 sha: 83926138aae9cd77e8c74ec8678b38417bebaa3d

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Latest

#### How is this defect reproduced ? 
mbed test -t GCC_ARM -m UBLOX_EVK_ODIN_W2 --compile
mbed test -t ARMC6 -m UBLOX_EVK_ODIN_W2 --compile
mbed test -t IAR -m UBLOX_EVK_ODIN_W2 --compile

0xc0170, 2020-03-31:
cc @ARMmbed/mbed-os-ipcore

0xc0170, 2020-03-31:
@michalpasztamobica You might know why is this failing?

michalpasztamobica, 2020-03-31:
I don't know at first glance, but I will check it.

kivaisan, 2020-03-31:
This came with latest `SocketAddress` modification. I can provide fixes.

@qasim-ublox , I cannot test IAR at the moment, so could you try if https://github.com/kivaisan/mbed-os/commit/ea2fb0a251a597e120f1a2901b1239d8baa6d616 fixes the `unique_ptr::reset` issue with IAR?

kivaisan, 2020-03-31:
I also made fixes for UT. @qasim-ublox could you please test PR #12731?

0xc0170, 2020-03-31:
Thanks @qasim-ublox for reporting this regression on master. We had found the same issue in the recent nightly run today.

qasim-ublox, 2020-03-31:
I manually added changes in my repo but still getting error. However, now the error is different.

### GCC:
UBLOX_EVK_ODIN_W2::GCC_ARM::MBED-OS-TESTS-NETWORK-MULTIHOMING
        Building project multihoming (UBLOX_EVK_ODIN_W2, GCC_ARM)
        Scan: GCC_ARM
        Scan: multihoming
        Compile [ 25.0%]: main.cpp
        Compile [ 50.0%]: multihoming_asynchronous_dns.cpp
        Compile [ 75.0%]: multihoming_synchronous_dns.cpp
        Compile [100.0%]: multihoming_udpsocket_echotest.cpp
        [Error] mbed_trace.h@136,66: 'TRACE_GROUP' was not declared in this scope
        [Error] mbed_trace.h@136,66: 'TRACE_GROUP' was not declared in this scope
        [Error] mbed_trace.h@136,66: 'TRACE_GROUP' was not declared in this scope
        [Error] mbed_trace.h@129,66: 'TRACE_GROUP' was not declared in this scope

### IAR:
Compile [ 69.7%]: SocketAddress.cpp
[Error] memory@1650,12: [Pe300]: a pointer to a bound function may only be used to call the function
[Error] memory@1651,0: [Pe300]: a pointer to a bound function may only be used to call the function
[ERROR]
      pointer _Old = this->_Myptr;
                           ^
"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.2\arm\inc\cpp\memory",1650  Error[Pe300]: a pointer to a bound function may only be used to call the function
          detected during instantiation of "inline __interwork __softfp void std::unique_ptr<_Ty [], _Dx>::reset(nullptr_t) [with _Ty=char, _Dx=std::default_delete<char []>]" at line 47 of ".\mbed-os\features\netsocket\SocketAddress.cpp"

      this->_Myptr = _Ptr;
            ^
"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.2\arm\inc\cpp\memory",1651  Error[Pe300]: a pointer to a bound function may only be used to call the function
          detected during instantiation of "inline __interwork __softfp void std::unique_ptr<_Ty [], _Dx>::reset(nullptr_t) [with _Ty=char, _Dx=std::default_delete<char []>]" at line 47 of ".\mbed-os\features\netsocket\SocketAddress.cpp"

Here is my diff:
[diff.txt](https://github.com/ARMmbed/mbed-os/files/4410342/diff.txt)

kjbracey-arm, 2020-04-01:
Oh dear, it looks like there's another bug in that library function. It should read `this->_Myptr() = _Ptr;`

Try another form - change the line in `SocketAddress` to

    _ip_address = nullptr;

kivaisan, 2020-04-01:
I also pushed TRACE_GROUP fix into fix PR. @qasim-ublox could you try Kevin's suggestion? If it works, I can include it in the fix PR as well.

qasim-ublox, 2020-04-01:
Both fixes are working .Logs are attached for reference.
[iar_mbed-os-example-wifi_build_logs.txt](https://github.com/ARMmbed/mbed-os/files/4414006/iar_mbed-os-example-wifi_build_logs.txt)
[gcc_build_logs.txt](https://github.com/ARMmbed/mbed-os/files/4414014/gcc_build_logs.txt)

kivaisan, 2020-04-01:
@qasim-ublox Thank you for testing. I also added IAR fix to the PR so if you could add your review into it.

MarceloSalazar, 2020-04-02:
@qasim-ublox I'd suggest we use and test 5.15 branch as there is where we'd like to keep support for the ODIN W2 platform.
Any PR/fixes for the ODIN should be raised against the 5.15 branch.
As we agreed, master (path to Mbed 6) won't have support for ODIN.

qasim-ublox, 2020-04-02:
@aqib-ublox

se7ensong, 2020-04-06:
@MarceloSalazar , will mbed-os 6 support ODIN once released? Thank you.

MarceloSalazar, 2020-04-06:
> will mbed-os 6 support ODIN

I'm afraid it won't, but we continue to work with @ARMmbed/team-ublox on the 5.15 branch

MarceloSalazar, 2020-04-08:
@kivaisan can you please confirm the fixes are in the 5.15 branch?
If not, please re-raise them against 5.15.
Note code freeze is tomorrow. Thanks!

kivaisan, 2020-04-09:
@MarceloSalazar I think it was agreed not to fix those few small test case issues in 5.15 or at least nobody requested those. Quote from PR:

> 
> > Note we should be applying these changes to the 5.15 branch too, as that's where we'll continue to support the Ublox ODIN platform.
> 
> Not necessarily:
> 
> - TEST_ASSERT_EQUAL actually does not check the correct thing, but is this so critical that should be fixed in 5.15. I leave it to maintainers to decide.
> - TRACE_GROUP build issue occurs only when mbed trace is enabled. By default traces are not enabled for tests. So I don't see this that critical at all.
> - unique_ptr::reset workaround is a regression from mbed-os 6 SocketAddress refactoring. Therefore does not apply to mbed-os 5.15.
>

kivaisan, 2020-04-09:
@MarceloSalazar And I just checked that actually none of the issues exist in 5.15. First check is new in mbed-os 6. And 5.15 version of the test does not use mbed-trace.

0xc0170, 2020-04-09:
> @MarceloSalazar And I just checked that actually none of the issues exist in 5.15. First check is new in mbed-os 6. And 5.15 version of the test does not use mbed-trace.

@qasim-ublox it means this could be closed as "not supported" anymore (as on master, so nothing to fix).

PR #12731 (2020-04-15): Testing: Fix multihoming test compilation issues

### Summary of changes 

Fix multihoming test compilation issues.

- TEST_ASSERT_EQUAL compares int values and converting SocketAddress as int is not supported.
- TRACE_GROUP was missing preventing trace builds
- Workaround for SocketAddress `_ip_address.reset()` compilation problem with IAR.

Fixes #12720 

#### Impact of changes 

##
closed 2020-04-09 by 0xc0170