ARMmbed/mbed-os #12732: FUTURE_SEQUANA: failing to build on master

0xc0170, 2020-03-31:
### Description of defect

The logs for the latest CI nightly:

```
[Error] stdio_init.cpp@29,1: unknown type name 'Serial'
[ERROR] ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6_FUTURE/TARGET_FUTURE_SEQUANA/stdio_init.cpp:29:1: error: unknown type name 'Serial'
Serial _stdio_uart_object(STDIO_UART_TX, STDIO_UART_RX);
^
1 error generated.
```

`stdio_init.cpp` file needs an update. cc @ARMmbed/team-cypress 

#### Target(s) affected by this defect ?

FUTURE_SEQUANA

#### Toolchain(s) (name and version) displaying this defect ?

ARM and GCC ARM

#### What version of Mbed-os are you using (tag or sha) ?

dba3962f1699380d5d706f0dc0ad73083bb5d57e

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Same as in CI (latest)

#### How is this defect reproduced ? 

Build blinky

kyle-cypress, 2020-03-31:
@lrusinowicz , are you the right person to update TARGET_FUTURE_SEQUANA?

0xc0170, 2020-04-01:
Looks like it is being removed, will reference this issue in the PR now

0xc0170, 2020-04-02:
PR #12738 is removing this one, I'll close this one as resolved.

closed 2020-04-02 by 0xc0170