ARMmbed/mbed-os #12760: ATHandler::read_string considers comma as separator inside quoted string

vznncv, 2020-04-05:
### Description of defect

`ATHandler` parser of the cellular devices doesn't process quoted string correctly.
`ATHandler::read_string` method considers a comma as separator inside quoted string and don't consume part of the string after the comma.

For example if ATHandler gets input data:
```
+CCLK: "20/04/05,15:38:57+12"

OK
```
then the code:
```
at.resp_start("+CCLK:");
read_res = at.read_string(buf, buf_size);
at.resp_stop();
```
will read only "20/04/05" instead of "20/04/05,15:38:57+12"

#### Target(s) affected by this defect ?

Any

#### Toolchain(s) (name and version) displaying this defect ?

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not relevant

#### How is this defect reproduced ? 

Run the following example:
```
#include "ATHandler.h"
#include "mbed.h"
#include "string.h"

class StringBufFileHandle : public FileHandle {
private:
    const char *_str;
    off_t _pos;
    off_t _len;

public:
    StringBufFileHandle(const char *str)
        : _str(str)
        , _pos(0)
        , _len(strlen(str))
    {
    }
    ~StringBufFileHandle(){};

    // FileHandle API implementation
    short poll(short events) const override
    {
        return POLLIN;
    }

    // FileHandle interface
    ssize_t read(void *buffer, size_t size) override
    {
        size_t unread_bytes = _len - _pos;
        size_t res = size < unread_bytes ? size : unread_bytes;
        memmove(buffer, _str + _pos, res);
        _pos += res;
        return res;
    }
    ssize_t write(const void *buffer, size_t size) override
    {
        return -1;
    }
    off_t seek(off_t offset, int whence) override
    {
        off_t new_pos;
        switch (whence) {
        case SEEK_SET:
            new_pos = offset;
            break;
        case SEEK_CUR:
            new_pos += offset;
            break;
        case SEEK_END:
            new_pos = _len + offset;
            break;
        default:
            return -1;
        }
        if (new_pos < 0 || new_pos > _len) {
            return -1;
        }
        _pos = new_pos;
        return _pos;
    }
    int close() override
    {
        return 0;
    }
};

int main()
{
    const size_t buf_size = 32;
    char buf[buf_size];
    int err;
    ssize_t read_res;

    StringBufFileHandle fh("+CCLK: \"20/04/05,15:38:57+12\"\r\n\r\nOK\r\n");
    ATHandler at(&fh, *mbed_event_queue(), 1000, CRLF);

    at.lock();
    at.resp_start("+CCLK:");
    read_res = at.read_string(buf, buf_size);
    at.resp_stop();
    err = at.unlock_return_error();

    printf("ATHandler read string demo:\n");
    printf("err: %i\n", err);
    printf("read_res: %i\n", read_res);
    printf("buf: %s\n", buf);

    while (true) {
        ThisThread::sleep_for(1000);
    }
}
```

expected output:

```
ATHandler read string demo:
err: 0
read_res: 20
buf: 20/04/05,15:38:57+12
```

actual output:

```
ATHandler read string demo:
err: 0
read_res: 8
buf: 20/04/05
```

0xc0170, 2020-04-06:
cc @ARMmbed/mbed-os-wan

0xc0170, 2020-04-09:
Will be fixed via #12763

PR #12763 (2020-04-15): Cellular: Fix ATHandler::read_string to handle delimiter inside string

### Summary of changes 

For example if input buffer includes:
`+CCLK: "20/04/05,15:38:57+12"`

`read_string` return `"20/04/05,15:38:57+12"` instead of `"20/04/05"`

Fixes #12760 

#### Impact of changes 

#### Migration actions required
closed 2020-04-15 by 0xc0170