ARMmbed/mbed-os #12764: Build failure seen with ARMC6 compile

mmahadevan108, 2020-04-06:
### Description of defect
The following error was seen when compiling the master branch with ARMC6 toolchain

[Error] @0,0: L6218E: Undefined symbol EthernetInterface::~EthernetInterface() (referred from BUILD/tests/MIMXRT1050_EVK/ARMC6/mbed-os/features/netsocket/EthernetInterface.o).
[Error] @0,0: L6218E: Undefined symbol thunk{-124} to EthernetInterface::~EthernetInterface() (referred from BUILD/tests/MIMXRT1050_EVK/ARMC6/mbed-os/features/netsocket/EthernetInterface.o)

No errors seen when compiling with GCC_ARM toolchain.

#### Target(s) affected by this defect ?
MIMXRT1050, K64F

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6

#### What version of Mbed-os are you using (tag or sha) ?
'mbed-os-99.99.99
    
sha: 4b3cddff2a66d5b13ad861e8782d1d630c2479c1

-->

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
Compile with ARMC6 toolchain K64F

0xc0170, 2020-04-09:
I've seen this before. It is because of older ARMCC 6 version. Please update to 6.13, should be all fine there.

0xc0170, 2020-04-09:
I'll close this as resolved. If anything else outstanding, please reopen

mmahadevan108, 2020-04-09:
Below is what I have running on my machine:
_______________________
Tool Version Numbers:
Toolchain:        MDK Professional 5.29 (Flex): 40 user(s)  Version: 5.29.0.0
Toolchain Path:    C:\Keil_v5\ARM\ARMCLANG\Bin
C Compiler:         ArmClang.exe        V6.13.1
Assembler:          Armasm.exe        V6.13.1
Linker/Locator:     ArmLink.exe        V6.13.1
Library Manager:    ArmAr.exe        V6.13.1
Hex Converter:      FromElf.exe        V6.13.1
CPU DLL:               SARMCM3.DLL          V5.29.0.0
Dialog DLL:         TCM.DLL              V1.36.5.0
Target DLL:             CMSIS_AGDI.dll       V1.32.0.0
Dialog DLL:         TCM.DLL              V1.36.5.0
______________________________

Looks like 6.13. The issue is occuring due the below commit 1ecd9604f62a20ec68653bd8bfc2713b71886440

0xc0170, 2020-04-09:
There is a mention in PR:

> This option appears to exist in ARM Compiler 6.11, but generates an apparently spurious linker error about EthernetInterface. It works in ARM Compiler 6.13, so this PR needs to wait until the compiler is updated.

your version should be up to date. I haven't seen this locally or in CIs. I'll reopen to investigate.

cc @kjbracey-arm

mmahadevan108, 2020-04-09:
I had an error in my setup, I was using version 6.11. I have updated to 6.14, issue was resolved.

closed 2020-04-09 by mmahadevan108