ARMmbed/mbed-os #12773: call_every seems factor 10 too slow.

Reissner, 2020-04-08:
I use lpc1768 with newest firmware, os 5.15.1, GCC9, mbed-cli 1.10.2. 

Code: 
```
DigitalOut ledHeartBeat(LED4);

EventQueue evntQueue;

void toggleLedHeartBeat(void) {
  ledHeartBeat = !ledHeartBeat;
}

int main(void) {
  evntQueue.call_every(100, callback(toggleLedHeartBeat));
  evntQueue.dispatch();
}
```

but led inverts every second... i.e. 1000ms. 
According to documentation 100 is interpreted as ms. 
What is wrong here? 

### Description of defect

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

kjbracey-arm, 2020-04-08:
Someone else had the same problem, and it was apparently solved by a firmware update: https://github.com/ARMmbed/mbed-os/issues/12347

Not quite sure why though, and you say you have the newest firmware. Can you check via the link on that issue?

adbridge, 2020-04-08:
@Reissner looks like you added the description text in the wrong place,  I will fix for you

0xc0170, 2020-04-14:
> Someone else had the same problem, and it was apparently solved by a firmware update: #12347

@Reissner have you checked the firmware if that resolved this or not?

0xc0170, 2020-04-27:
Closing this issue as its been almost 3 weeks without any update.

closed 2020-04-27 by 0xc0170