ARMmbed/mbed-os #12782: NACK is received sometimes sporadically while reading the onboard sensor data from B-L475E-IOT01A2 board

iampoonam, 2020-04-09:
### Description of defect
There are no  blocking mode functions in i2c_api.c for I2C read and write that call STM  HAL functions like HAL_I2C_MASTER_TRANSMIT() and HAL_I2C_MASTER_RECEIVE().
With the current i2c_write and i2c_read function in i2c_api.c , NACK is received sometimes sporadically while reading the onboard sensor data from B-L475E-IOT01A2 board

#### Target(s) affected by this defect ?
B-L475E-IOT01A2 board

#### What version of Mbed-os are you using?
MBED-5.12.0

jeromecoutant, 2020-04-09:
> #### What version of Mbed-os are you using?
> MBED-5.12.0

Hi
Could you check master branch ?
Thx

0xc0170, 2020-04-27:
@iampoonam Please review the comments. Was this fixed in the latest release (if yes, you can cherry-pick the change) or update.

jeromecoutant, 2020-06-08:
@0xc0170 maybe we can close this one ?

0xc0170, 2020-08-31:
I'll close this as there are too many reminders. Please update the first comment , leaving the issue template as it should be and fill in details (we process them internally, they are mandatory). Thanks for understanding.

closed 2020-08-31 by 0xc0170