ARMmbed/mbed-os #12784: QSPIFBlockDevice: unable to set QE bit on ISSI Flash memories

Uruloke, 2020-04-09:
### Description of defect

While trying to activate LittleFileSystem on a ISSI IS25LP128 Flash memory I noticed it failed every time when trying to set the QE bit (quad enable) in QSPIFBlockDevice. I did some digging into this and it appears this happens because by default QSPIFBlockDevice assumes a Flash memory have two Status registers and when writing the Status registers it writes 2 bytes using the WRSR instruction. This will not work on a ISSI memory and will cause it to be unable to set the QE bit.

ISSI datasheet specifies that the WRSR instruction only write a single byte.

I fixed this issue by setting _num_status_registers = 1 for vendor id 0x9d and disabling reads of status register 2 if _num_status_registers is 1. See attached patch file.

[QSPIFBlockDevice-support-issi-devices.patch.txt](https://github.com/ARMmbed/mbed-os/files/4456133/QSPIFBlockDevice-support-issi-devices.patch.txt)

#### Target(s) affected by this defect ?

All targets using an ISSI Flash memory part.

#### Toolchain(s) (name and version) displaying this defect ?

Embedded ARM GCC 9 2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.8.3

#### How is this defect reproduced ? 

Connect an ISSI Flash memory device to a target and attempt to initialize LittleFileSystem on a QSPIFBlockDevice using the device.

0xc0170, 2020-04-15:
cc @ARMmbed/mbed-os-storage

VeijoPesonen, 2020-05-07:
@Uruloke would you please create a PR based on your patch. [Instructions](https://os.mbed.com/docs/mbed-os/v5.15/contributing/workflow.html#contributions) for contributing an enhancement.

Uruloke, 2020-05-08:
Sure, I can try. :)

Uruloke, 2020-05-08:
@VeijoPesonen Okay, I created a PR. I think I did everything correctly!

Uruloke, 2020-05-08:
@VeijoPesonen Created a PR against master as you requested.

VeijoPesonen, 2020-05-11:
@Uruloke Great. Thank you for the PR. Keep them coming :)

PR #12947 (2020-05-12): Fix setting the QE bit on ISSI Flash memories

### Summary of changes 

By default QSPIFBlockDevice assumes a Flash memory have two Status registers and when writing the Status registers it writes 2 bytes using the WRSR instruction. This will not work on a ISSI Flash memory as they only have a single status register and attempting to write two bytes using WRSR will cause it to ignore the request and be unable to set the QE bit.

ISSI datasheet specifies that the WRSR instruction only write a single byte so override the number of status registers for ISSI Flash memories.

Fixes #12784.

#### Im [... cut]
closed 2020-05-12 by 0xc0170